tue-robotics/tue_robocup #10: [challenge_rips] Amigo didn't get to the exit

Rayman, 2015-01-12:
I repeated the RIPS 3x but only the first time the navigation to the exit failed. Below is the output of the executive. Is this a random failure?

```bash
$ rosrun challenge_rips rips.py amigo
...
[INFO] [WallTime: 1421077210.068279] State machine transitioning 'ASK_CONTINUE':'continue'-->'SAY_CONTINUEING'
[INFO] [WallTime: 1421077210.069525] 'I heard continue, so I will move to the exit now. See you guys later!'
[INFO] [WallTime: 1421077210.076467] State machine transitioning 'SAY_CONTINUEING':'spoken'-->'GO_TO_EXIT'
[INFO] [WallTime: 1421077210.077914] State machine starting in initial state 'START_ANALYSIS' with userdata: 
	[]
[INFO] [WallTime: 1421077210.084270] State machine transitioning 'START_ANALYSIS':'done'-->'MONITORED_NAVIGATE'
[INFO] [WallTime: 1421077210.085587] Concurrence starting with userdata: 
	[]
[INFO] [WallTime: 1421077210.087945] State machine starting in initial state 'GET_PLAN' with userdata: 
	[]
[INFO] [WallTime: 1421077210.170921] State machine transitioning 'GET_PLAN':'unreachable'-->'RESET_WORLD_MODEL'
[ERROR] [WallTime: 1421077210.306769] Received comm state PREEMPTING when in simple state DONE with SimpleActionClient in NS /cb_base_navigation/local_planner_interface/action_server
[INFO] [WallTime: 1421077212.821973] State machine terminating 'RESET_WORLD_MODEL':'succeeded':'unreachable'
[INFO] [WallTime: 1421077212.822947] Concurrent state 'NAVIGATE' returned outcome 'unreachable' on termination.
[INFO] [WallTime: 1421077213.093062] Concurrent state 'MONITOR' returned outcome 'passed' on termination.
[INFO] [WallTime: 1421077213.093621] Concurrent Outcomes: {'NAVIGATE': 'unreachable', 'MONITOR': 'passed'}
[INFO] [WallTime: 1421077213.093927] State machine transitioning 'MONITORED_NAVIGATE':'unreachable'-->'STOP_ANALYSIS_UNREACHABLE'
[INFO] [WallTime: 1421077213.098102] 

Navigation summary:
Covered 1.03355603125e-05 meters in 3.01413798332 seconds (3.429026929e-06) m/s avg.
Result = unreachable with 0 plans, 0 clears and 0 resets

[INFO] [WallTime: 1421077213.098478] State machine terminating 'STOP_ANALYSIS_UNREACHABLE':'done':'unreachable'
[INFO] [WallTime: 1421077213.098732] State machine transitioning 'GO_TO_EXIT':'unreachable'-->'AT_END'
[INFO] [WallTime: 1421077213.099092] 'Goodbye'
[INFO] [WallTime: 1421077217.863622] State machine terminating 'AT_END':'spoken':'Done'
Final outcome: Done
[INFO] [WallTime: 1421077217.864715] Torso cancelling all goals on close
[INFO] [WallTime: 1421077217.865801] left arm cancelling all goals on all arm-related ACs on close
[INFO] [WallTime: 1421077217.867004] right arm cancelling all goals on all arm-related ACs on close
```

jlunenburg, 2015-01-21:
This is definitely not enough information, with this output I don't have a clue why it says the exit is unreachable...

rokusottervanger, 2015-04-02:
If it's not specific enough and not possible to reproduce, can this issue be closed?

closed 2015-04-02 by Rayman