tue-robotics/tue_robocup #13: [basic_functionalities] Amigo didn't pick and place

rokusottervanger, 2015-01-13:
In the pick and place state, Amigo did not get close enough to the table to pick or place an object.

rokusottervanger, 2015-02-17:
This was probably just because the right arm wasn't available for this. I'm going to close this issue as this problem is already addressed by tue-robotics/robot_smach_states#11, which is fixed

LoyVanBeek, 2015-02-18:
@rokusottervanger there is an ArmDesignator and some subclasses. 
They check whether an arm is available for some task (i.e. is not holding an item or *is* holding a specific item.)
There currently is no check for an arm being available/attached/working AFAIK. Adding such a property of an Arm-object would be step 1. For example, you could check that there are joint measurements being received for some arm.

rokusottervanger, 2015-02-18:
I see. I will reopen this then...

Rayman, 2015-03-27:
We now have an `occupied_by` and a `operational` property. Does the pick and place work?

closed 2015-08-07 by Rayman