tue-robotics/tue_robocup #19: [robo_zoo] qr_detector_zbar is now an ed_perception plugin?

LoyVanBeek, 2015-01-27:
RoboZoo needs QR readings to see which aaction to do. Previously, a node published this on a topic but now we use ed :question:  :confused:

LoyVanBeek, 2015-01-27:
Aha, its in http://roboticssrv.wtb.tue.nl/svn/ros/code/tue-ros-pkg/trunk/tue_perception/qr_detector_zbar/ and not yet a catkinized package. TODO next week.

Rayman, 2015-02-04:
There is also a qr detector plugin in ed [plugins/qr_detector.h](https://github.com/tue-robotics/ed_perception/blob/f993443af0a54071cd3f5581872b91e9263cff5e/plugins/qr_detector.h). Maybe we could interface that?

LFBFerreira, 2015-02-04:
Should be easy to create a new perception plugin with the code that is on qr_detector_zbar.cpp. Think the last person that used this was @reinzor for loading a table model from a sticker on top of the table.

If ED can segment an object with a QR code you should have no problem

reinzor, 2015-04-21:
Fixed separate node , so not integrated in ED, this is not useful ..

closed 2015-04-21 by reinzor