tue-robotics/tue_robocup #22: Navigation crashes

jlunenburg, 2015-02-24:
@LoyVanBeek: might this have anything to do with recent changes regarding the resolution of designators?

[ERROR] [WallTime: 1424764958.627144] InvalidUserCodeError: Could not execute state 'GET_PLAN' of type '<robot_smach_states.navigation.navigation.getPlan object at 0x40e61d0>': Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/smach/state_machine.py", line 241, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/hydro/system/src/robot_smach_states/src/robot_smach_states/navigation/navigation.py", line 72, in execute
    constraint = self.constraint_function()
  File "/home/amigo/ros/hydro/system/src/robot_smach_states/src/robot_smach_states/navigation/navigate_to_waypoint.py", line 30, in generateConstraint
    pose = e.data["pose"]
AttributeError: 'str' object has no attribute 'data'

[ERROR] [WallTime: 1424764958.628238] InvalidStateError: Could not execute child state 'NAVIGATE': Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/smach/concurrence.py", line 326, in _state_runner
    self._remappings[label]))
  File "/opt/ros/hydro/lib/python2.7/dist-packages/smach/state_machine.py", line 347, in execute
    container_outcome = self._update_once()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/smach/state_machine.py", line 246, in _update_once
    raise smach.InvalidUserCodeError("Could not execute state '%s' of type '%s': " % ( self._current_label, self._current_state ) + traceback.format_exc())
InvalidUserCodeError: <smach.exceptions.InvalidUserCodeError instance at 0x4101a28>

LoyVanBeek, 2015-02-24:
I suspect, on a first guess, that NavigateToWaypoint received a designator that resolves to an entity ID instead of to an entity.
Maybe that went wrong with https://github.com/tue-robotics/robot_smach_states/issues/14

LoyVanBeek, 2015-03-27:
I'll take a look at this tomorrow

LoyVanBeek, 2015-04-01:
This has to do with the *parse* option to ed.get_entities. If that is False, the data is not loaded. And indeed, there is no such attribute. 

    pose = e.data["pose"]
    AttributeError: 'str' object has no attribute 'data'

Do you have this problem still and what designator/waypoint does it happen wih?

closed 2015-04-13 by LoyVanBeek