tue-robotics/tue_robocup #23: Crash manipulation challenge

reinzor, 2015-02-26:
amigo@amigo3 $ rosrun challenge_manipulation manipulation.py amigo
[INFO] [WallTime: 1424974943.510761] Using robot 'amigo'.
If you want to test an executive, use the --test option and specify what parts you want to mock.
[WARN] [WallTime: 1424974943.525109] Nav_analyser: Bagging = False
[INFO] [WallTime: 1424974943.526710] Waiting for the global planner services ...
[WARN] [WallTime: 1424974944.594158] Cannot find gripper left server
[WARN] [WallTime: 1424974945.662472] Cannot find gripper right server
[INFO] [WallTime: 1424974945.681755] Waiting for service /amigo/text_to_speech/speak
[INFO] [WallTime: 1424974945.685810] Initializing ebutton listener
EdEntityDesignator(robot, type=, center_point=x: 0.0 y: 0.0 z: 0.0, radius=0, id=plastic_cabinet, parse=True, criteriafuncs=[]) = bookcase
EdEntityDesignator(robot, type=, center_point=header:    seq: 0   stamp:      secs: 1424974945     nsecs: 695074081   frame_id: /plastic_cabinet point:    x: 0   y: 0   z: 0, radius=0.75, id=, parse=True, criteriafuncs=[<function <lambda> at 0x7f7f64011230>, <function <lambda> at 0x7f7f640112a8>, <function inside at 0x7f7f64011320>]) = current_item
<__main__.EmptySpotDesignator object at 0x7f7f64010e90> = place_position
<robot_smach_states.util.designators.designator.UnoccupiedArmDesignator object at 0x7f7f64010ed0> = empty_arm_designator
<robot_smach_states.util.designators.designator.ArmHoldingEntityDesignator object at 0x7f7f64010f10> = arm_with_item_designator
[INFO] [WallTime: 1424974945.739567] State machine starting in initial state 'AWAIT_START' with userdata: 
	[]
[INFO] [WallTime: 1424974945.740789] State machine starting in initial state 'SAY' with userdata: 
	[]
[INFO] [WallTime: 1424974945.741818] 'I will continue my task if you say continue.'
[INFO] [WallTime: 1424974950.129080] State machine transitioning 'SAY':'spoken'-->'HEAR'
[INFO] [WallTime: 1424974952.390056] State machine terminating 'HEAR':'heard':'continue'
[INFO] [WallTime: 1424974952.390996] State machine transitioning 'AWAIT_START':'continue'-->'NAV_TO_START'
[INFO] [WallTime: 1424974952.392205] State machine starting in initial state 'START_ANALYSIS' with userdata: 
	[]
[INFO] [WallTime: 1424974952.398508] State machine transitioning 'START_ANALYSIS':'done'-->'RESET_SM_NAV'
[INFO] [WallTime: 1424974952.399660] State machine transitioning 'RESET_SM_NAV':'done'-->'MONITORED_NAVIGATE'
[INFO] [WallTime: 1424974952.400688] Concurrence starting with userdata: 
	[]
[INFO] [WallTime: 1424974952.402455] State machine starting in initial state 'GET_PLAN' with userdata: 
	[]
[ERROR] [WallTime: 1424974952.550457] Called get_state when no goal is running
[INFO] [WallTime: 1424974952.551861] State machine transitioning 'GET_PLAN':'unreachable'-->'RESET_WORLD_MODEL'
[INFO] [WallTime: 1424974952.553038] State machine terminating 'RESET_WORLD_MODEL':'succeeded':'unreachable'
[INFO] [WallTime: 1424974952.553788] Concurrent state 'NAVIGATE' returned outcome 'unreachable' on termination.
[WARN] [WallTime: 1424974952.905661] Starting breakout check
[INFO] [WallTime: 1424974952.907006] Concurrent state 'MONITOR' returned outcome 'passed' on termination.
[INFO] [WallTime: 1424974952.908184] Concurrent Outcomes: {'NAVIGATE': 'unreachable', 'MONITOR': 'passed'}
[INFO] [WallTime: 1424974952.909055] State machine transitioning 'MONITORED_NAVIGATE':'unreachable'-->'STOP_ANALYSIS_UNREACHABLE'
[INFO] [WallTime: 1424974952.915879] 

Navigation summary:
Covered 0.0 meters in 0.517087936401 seconds (0.0) m/s avg.
Result = unreachable with 0 plans, 0 clears and 0 resets

[INFO] [WallTime: 1424974952.917001] State machine terminating 'STOP_ANALYSIS_UNREACHABLE':'done':'unreachable'
[INFO] [WallTime: 1424974952.917782] State machine transitioning 'NAV_TO_START':'unreachable'-->'RANGE_ITERATOR'
[INFO] [WallTime: 1424974952.918791] Iterating 0 of <function <lambda> at 0x7f7f640111b8>
[INFO] [WallTime: 1424974952.919700] State machine starting in initial state 'NAV_TO_OBSERVE_BOOKCASE' with userdata: 
	[]
[INFO] [WallTime: 1424974952.920588] State machine starting in initial state 'START_ANALYSIS' with userdata: 
	[]
[INFO] [WallTime: 1424974952.926746] State machine transitioning 'START_ANALYSIS':'done'-->'RESET_SM_NAV'
[INFO] [WallTime: 1424974952.927968] State machine transitioning 'RESET_SM_NAV':'done'-->'MONITORED_NAVIGATE'
[INFO] [WallTime: 1424974952.928964] Concurrence starting with userdata: 
	[]
[INFO] [WallTime: 1424974952.931275] State machine starting in initial state 'GET_PLAN' with userdata: 
	[]
[INFO] [WallTime: 1424974953.099797] Goal handle = <actionlib.action_client.ClientGoalHandle instance at 0x7f7eb0032098>
[INFO] [WallTime: 1424974953.101381] 'Getting there!'
[INFO] [WallTime: 1424974953.106662] State machine transitioning 'GET_PLAN':'goal_ok'-->'EXECUTE_PLAN'
[WARN] [WallTime: 1424974953.432760] Starting breakout check
[WARN] [WallTime: 1424974954.115778] Executing alternative path!
[INFO] [WallTime: 1424974954.118433] Goal handle = <actionlib.action_client.ClientGoalHandle instance at 0x7f7eb0034680>
[WARN] [WallTime: 1424974962.860800] Breaking out for the first time
[INFO] [WallTime: 1424974962.861710] Concurrent state 'MONITOR' returned outcome 'preempted' on termination.
[INFO] [WallTime: 1424974963.189491] execute: preempt_requested
[INFO] [WallTime: 1424974963.190664] State machine terminating 'EXECUTE_PLAN':'preempted':'preempted'
[INFO] [WallTime: 1424974963.192043] Concurrent state 'NAVIGATE' returned outcome 'preempted' on termination.
[INFO] [WallTime: 1424974963.193820] Concurrent Outcomes: {'NAVIGATE': 'preempted', 'MONITOR': 'preempted'}
[INFO] [WallTime: 1424974963.194809] State machine transitioning 'MONITORED_NAVIGATE':'arrived'-->'STOP_ANALYSIS_SUCCEED'
[... 31 lines cut ...]
    kwargs_resolved = {key:value.resolve() for key,value in self.kwargs.iteritems()}
  File "/home/amigo/ros/hydro/system/src/challenge_manipulation/src/manipulation.py", line 46, in <dictcomp>
    kwargs_resolved = {key:value.resolve() for key,value in self.kwargs.iteritems()}
  File "/home/amigo/ros/hydro/system/src/challenge_manipulation/src/manipulation.py", line 56, in resolve
    entity = super(EntityNotOnListDesignator, self).resolve()
  File "/home/amigo/ros/hydro/system/src/robot_smach_states/src/robot_smach_states/util/designators/designator.py", line 217, in resolve
    "No entities found in {0}".format(self))
DesignatorResolvementError: No entities found in EdEntityDesignator(robot, type=, center_point=header:    seq: 0   stamp:      secs: 1424974945     nsecs: 695074081   frame_id: /plastic_cabinet point:    x: 0   y: 0   z: 0, radius=0.75, id=, parse=True, criteriafuncs=[<function <lambda> at 0x7f7f64011230>, <function <lambda> at 0x7f7f640112a8>, <function inside at 0x7f7f64011320>])

[ERROR] [WallTime: 1424974967.899399] Could not execute Iterator state 'SINGLE_ITEM'
[ERROR] [WallTime: 1424974967.900343] State 'RANGE_ITERATOR' failed to execute.
[ERROR] [WallTime: 1424974967.901473] Robot exited with smach.exceptions.InvalidUserCodeError,<smach.exceptions.InvalidUserCodeError instance at 0x7f7eb002cea8>,<traceback object at 0x7f7f6403a8c0>
[INFO] [WallTime: 1424974967.902610] Torso cancelling all goals on close
[INFO] [WallTime: 1424974967.903440] left arm cancelling all goals on all arm-related ACs on close
[INFO] [WallTime: 1424974967.904259] right arm cancelling all goals on all arm-related ACs on close
Traceback (most recent call last):
  File "/home/amigo/ros/hydro/system/src/challenge_manipulation/src/manipulation.py", line 256, in <module>
    startup(setup_statemachine, robot_name=robot_name)
  File "/home/amigo/ros/hydro/system/src/robot_smach_states/src/robot_smach_states/util/startup.py", line 69, in startup
    outcome = executioner.execute()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/smach/state_machine.py", line 347, in execute
    container_outcome = self._update_once()
  File "/opt/ros/hydro/lib/python2.7/dist-packages/smach/state_machine.py", line 244, in _update_once
    raise ex
smach.exceptions.InvalidUserCodeError: <smach.exceptions.InvalidUserCodeError instance at 0x7f7eb002cea8>

Rayman, 2015-02-27:
So apparently a `DesignatorResolvementError` crashes the whole state machine?
```
DesignatorResolvementError: No entities found in EdEntityDesignator(robot, type=, center_point=header: seq: 0 stamp: secs: 1424974945 nsecs: 695074081 frame_id: /plastic_cabinet point: x: 0 y: 0 z: 0, radius=0.75, id=, parse=True, criteriafuncs=[ at 0x7f7f64011230>, at 0x7f7f640112a8>, ])```

LoyVanBeek, 2015-02-28:
They do, unless you catch the exceptions...

LoyVanBeek, 2015-03-11:
No crashes yesterday. Main problem now is pre-grasp planning

closed 2015-03-11 by LoyVanBeek