tue-robotics/tue_robocup #34: Allow designators to work with dynamic input

egeerts, 2015-03-10:


LoyVanBeek, 2015-03-12:
Hi Erik, I made https://github.com/tue-robotics/robot_smach_states/tree/feature/34_dynamic_input Could you test whether this works for you? 
Just roscd robot_smach_states; git checkout feature/34_dynamic_input

egeerts, 2015-03-17:
smach.StateMachine.add('ACTION_NAVIGATE_TO_ROOM',
                                states.NavigateToSymbolic(robot, 
                                    {EdEntityDesignator(robot, id=("room_"+str(robot.reasoner.query("action_info('1','1_locations_rooms',A)")[0]['A']))) : "in" }, 
                                    EdEntityDesignator(robot, id=("room_"+str(robot.reasoner.query("action_info('1','1_locations_rooms',A)")[0]['A'])))),
                                transitions={   'arrived'           :   'FINISHED_TASK',
                                                'unreachable'       :   'FINISHED_TASK',
                                                'goal_not_defined'  :   'FINISHED_TASK'}) 

Je ziet bij de id dat ik de reasoner al aanroep, terwijl ik dit pas wil als ik bij deze state kom, vandaar dat er nu meteen al een error is bij deze state. 
Weet je een oplossing waardoor ik de query kan meegeven zonder dat de error optreedt? 
Of dat ik dit:

id=("room_"+str(robot.reasoner.query("action_info('1','1_locations_rooms',A)")))

meegeef en de designator automatisch het eerste antwoord pakt? 

Date: Thu, 12 Mar 2015 13:46:28 -0700

LoyVanBeek, 2015-03-17:
Well, id can be a designator itself too. 
So, if you create a designator (some class with a .resolve() method) that returns 

    "room_"+str(robot.reasoner.query("action_info('1','1_locations_rooms',A)")) 

then you're done.

LoyVanBeek, 2015-03-17:
Please see https://github.com/tue-robotics/tue_robocup/commit/fa40c2cfa0b0219b1aa83f4e0b032f9a8e570a4c

LoyVanBeek, 2015-04-01:
Fixed in https://github.com/tue-robotics/robot_smach_states/commit/585e52ddf50c0830fbfbed0a30cc15bfa9a0e205#diff-2e276f7c146e0524dc460b1009f7e9b3R29 for EdEntityDesignator

closed 2015-04-01 by LoyVanBeek