tue-robotics/tue_robocup #39: Macarena does not work

MaxBaeten, 2015-03-20:
Macarena does not work, ends up at: guys can you help me I am stuck in the macarena.
I couldn't find any invalid joint goals in the challenge however.

High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.12.1; written and copyright by Michael Hipp and others
	free software (LGPL/GPL) without any warranty but with best wishes

Directory: /home/amigo/ros/hydro/repos/https:/github.com/tue-robotics/tue_robocup.git/challenge_robo_zoo/src/challenge_robo_zoo_simple/
Playing MPEG stream 1 of 1: macarena.mp3 ...
MPEG 1.0 layer III, 128 kbit/s, 44100 Hz joint-stereo
[INFO] [WallTime: 1426846158.618662] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846158.619839] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846158.984935] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846158.986191] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846159.621936] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846159.623129] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846159.987996] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846159.989164] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846160.625251] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846160.626414] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846160.990987] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846160.992472] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846161.628449] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846161.629717] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846161.994571] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846161.995754] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846162.631794] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846162.633352] Send torso goal [0.3], timeout = 0.0
[WARN] [WallTime: 1426846162.982558] Cannot reach joint goal trajectory: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs: 0
    frame_id: ''
  joint_names: ['shoulder_yaw_joint_right', 'shoulder_pitch_joint_right', 'shoulder_roll_joint_right', 'elbow_pitch_joint_right', 'elbow_roll_joint_right', 'wrist_pitch_joint_right', 'wrist_yaw_joint_right']
  points: 
    - 
      positions: [0.0, 1.5, 0.0, 0.0, 0.0, 0.0, 0.0]
      velocities: []
      accelerations: []
      effort: []
      time_from_start: 
        secs: 5
        nsecs: 0
path_tolerance: []
goal_tolerance: []
goal_time_tolerance: 
  secs: 0
  nsecs: 0
[INFO] [WallTime: 1426846162.983609] 'Guys, could you help me, i'm stuck in the maca-rena'
[INFO] [WallTime: 1426846162.997759] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846162.999108] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846163.635509] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846163.636671] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846164.001157] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846164.002401] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846164.638051] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846164.639353] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846165.022297] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846165.024019] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846165.641642] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846165.642884] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846166.026155] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846166.027600] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846166.644932] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846166.646174] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846167.029751] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846167.031298] Send torso goal [0.3], timeout = 0.0
[ERROR] [WallTime: 1426846167.458798] Arms did not reach goal,  need help
[INFO] [WallTime: 1426846167.460058] State machine transitioning 'MACARENA':'Done'-->'RESET_ALL'
[INFO] [WallTime: 1426846167.462039] Send left arm to jointcoords 
[positions: [-0.1, -0.2, 0.2, 0.8, 0.0, 0.0, 0.0]
velocities: []
accelerations: []
effort: []
time_from_start:

MaxBaeten, 2015-03-20:
And the same happens for the head shoulders knee and toes:

[INFO] [WallTime: 1426846584.017017] State machine transitioning 'SELECT_RANDOM':'HOOFD_SCHOUDERS_KNIE_TEEN'-->'HOOFD_SCHOUDERS_KNIE_TEEN'
Playing music: /home/amigo/ros/hydro/repos/https:/github.com/tue-robotics/tue_robocup.git/challenge_robo_zoo/src/challenge_robo_zoo_simple/Xuxa.mp3
[INFO] [WallTime: 1426846584.018270] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846584.020093] Send torso goal [0.3], timeout = 0.0
If the music keeps going somehow, its PID is: 7494
[INFO] [WallTime: 1426846584.039769] hoofd
[INFO] [WallTime: 1426846584.040806] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846584.043265] Send right arm to jointcoords 
[positions: [0.0, 1.4, 0.0, 2.1, 1.5, 0.5, 0.0]
velocities: []
accelerations: []
effort: []
time_from_start: 
  secs: 5
  nsecs: 0]
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.12.1; written and copyright by Michael Hipp and others
	free software (LGPL/GPL) without any warranty but with best wishes

Directory: /home/amigo/ros/hydro/repos/https:/github.com/tue-robotics/tue_robocup.git/challenge_robo_zoo/src/challenge_robo_zoo_simple/
Playing MPEG stream 1 of 1: Xuxa.mp3 ...
MPEG 1.0 layer III, 128 kbit/s, 44100 Hz joint-stereo
[INFO] [WallTime: 1426846584.990546] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846584.991302] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846585.008720] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846585.009450] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846585.032576] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846585.033254] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846585.992987] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846585.994872] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846586.010642] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846586.012422] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846586.034894] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846586.036981] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846586.997296] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846586.998485] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846587.014569] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846587.015687] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846587.039619] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846587.040780] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846588.000498] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846588.001699] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846588.017791] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846588.019008] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846588.042645] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846588.043920] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846589.003720] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846589.004991] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846589.021080] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846589.022618] Send torso goal [0.3], timeout = 0.0
[WARN] [WallTime: 1426846589.045602] Cannot reach joint goal trajectory: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs: 0
    frame_id: ''
  joint_names: ['shoulder_yaw_joint_right', 'shoulder_pitch_joint_right', 'shoulder_roll_joint_right', 'elbow_pitch_joint_right', 'elbow_roll_joint_right', 'wrist_pitch_joint_right', 'wrist_yaw_joint_right']
  points: 
    - 
      positions: [0.0, 1.4, 0.0, 2.1, 1.5, 0.5, 0.0]
      velocities: []
      accelerations: []
      effort: []
      time_from_start: 
        secs: 5
        nsecs: 0
path_tolerance: []
[... 39 lines cut ...]
[INFO] [WallTime: 1426846595.043789] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846595.044735] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846595.067427] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846595.068836] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846596.026136] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846596.027647] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846596.046468] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846596.047540] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846596.070547] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846596.071553] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846597.029780] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846597.031078] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846597.049574] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846597.050851] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846597.073775] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846597.075232] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846598.032641] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846598.033859] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846598.052902] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846598.054000] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846598.077239] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846598.079160] Send torso goal [0.3], timeout = 0.0
[INFO] [WallTime: 1426846599.035966] Send torso goal [0.4], timeout = 0.0
[INFO] [WallTime: 1426846599.037128] Send torso goal [0.3], timeout = 0.0
[ERROR] [WallTime: 1426846599.039024] Arms did not reach goal,  need help

LoyVanBeek, 2015-03-20:
The joint goals are the same as last year, so something changed. Dunno what though. 
They probably need tweaking and tuning

reinzor, 2015-03-25:
I will look at this saturday

Rayman, 2015-08-07:
Did it work on robocup?

MaxBaeten, 2015-09-16:
I think so, I will close this issue. I think our RoboZoo needs a thorough examination anyhow if we want to keep it like this. But it is not a prio right now so I'll close this issue for now.

closed 2015-09-16 by MaxBaeten