tue-robotics/tue_robocup #42: [robo_zoo] Crash: look_at_point() got an unexpected keyword argument 'keep_tracking'

svddries, 2015-03-30:
[ERROR] [WallTime: 1427715083.216260] InvalidUserCodeError: Could not execute state 'GANGNAM' of type '<gangnam_style.GangNamStyle object at 0x2faf090>': Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/smach/state_machine.py", line 241, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/hydro/repos/https:/github.com/tue-robotics/tue_robocup.git/challenge_robo_zoo/src/challenge_robo_zoo_simple/gangnam_style.py", line 124, in execute
    gangnam_style(self.robot)
  File "/home/amigo/ros/hydro/repos/https:/github.com/tue-robotics/tue_robocup.git/challenge_robo_zoo/src/challenge_robo_zoo_simple/gangnam_style.py", line 104, in gangnam_style
    motionB(robot)
  File "/home/amigo/ros/hydro/repos/https:/github.com/tue-robotics/tue_robocup.git/challenge_robo_zoo/src/challenge_robo_zoo_simple/gangnam_style.py", line 77, in motionB
    robot.head.look_at_point(msgs.PointStamped(0,0,0, frame_id="/amigo/grippoint_right"), keep_tracking=True, pan_vel=1.0, tilt_vel=1.0)
TypeError: look_at_point() got an unexpected keyword argument 'keep_tracking'

[ERROR] [WallTime: 1427715083.217327] Robot exited with smach.exceptions.InvalidUserCodeError,<smach.exceptions.InvalidUserCodeError instance at 0x7f95ec01bdd0>,<traceback object at 0x2fb3098>

LoyVanBeek, 2015-03-30:
Being able to keep tracking is important!

Rayman, 2015-04-01:
Related to tue-robotics/robot_skills#13

closed 2015-08-07 by Rayman