tue-robotics/tue_robocup #44: No location when resolving designator using reasoner.

egeerts, 2015-04-02:
Previously I found an answer, now I don't anymore. Loy any idea what changed?
Running gpsr (in the gpsr branche) caused this problem:

[INFO] [WallTime: 1428002544.778076] 'What can I do for you?'
[INFO] [WallTime: 1428002547.549663] 'Okay I will advance to the workshop find carmen and follow her'
advance to the workshop find carmen and follow her
choice_key =  1_locations_rooms
choice_value = ' workshop '
choice_key =  1_action
choice_value = ' advance '
choice_key =  2_person_women
choice_value = ' carmen '
choice_key =  2_action_person
choice_value = ' find '
choice_key =  3_action_follow_women
choice_value = ' follow_her '
[INFO] [WallTime: 1428002552.417499] State machine transitioning 'ASK_ACTION':'done'-->'QUERY_SPECIFIC_ACTION'
[]
1
action 2: choice_key =  A
action 2: choice_value =  1_locations_rooms
a
room_workshop
[INFO] [WallTime: 1428002552.436778] State machine transitioning 'QUERY_SPECIFIC_ACTION':'navigate_room'-->'1_ACTION_NAVIGATE_TO_LOCATION'
[INFO] [WallTime: 1428002552.437267] State machine starting in initial state 'START_ANALYSIS' with userdata: 
	[]
[INFO] [WallTime: 1428002552.439766] State machine transitioning 'START_ANALYSIS':'done'-->'NAVIGATE'
[INFO] [WallTime: 1428002552.440256] State machine starting in initial state 'GET_PLAN' with userdata: 
	[]
[ERROR] [WallTime: 1428002552.551838] No entities found in EdEntityDesignator(robot, type=, center_point=x: 0.0 y: 0.0 z: 0.0, radius=0, id=, parse=True, criteriafuncs=[])
[ERROR] [WallTime: 1428002552.552626] One or more of entity_designator_area_name_map could not be resolved and resolved to None: 'NoneType' object has no attribute 'id'
[WARN] [WallTime: 1428002552.552918] Invalid constraints given to getPlan()
[INFO] [WallTime: 1428002552.553255] State machine terminating 'GET_PLAN':'goal_not_defined':'goal_not_defined'
[INFO] [WallTime: 1428002552.553735] State machine transitioning 'NAVIGATE':'goal_not_defined'-->'ABORT_ANALYSIS_NOT_DEFINED'
[INFO] [WallTime: 1428002552.554393] State machine terminating 'ABORT_ANALYSIS_NOT_DEFINED':'done':'goal_not_defined'
[INFO] [WallTime: 1428002552.554996] State machine transitioning '1_ACTION_NAVIGATE_TO_LOCATION':'goal_not_defined'-->'QUERY_SPECIFIC_ACTION'
[{'A': '1'}]
2

LoyVanBeek, 2015-04-02:
What changed: a failing designator resolves to none, and this is indeed what happens as you can see in de the error log. 
I can also see that the problem is one of the designators in the dictionary passed to  NavigateToSymbolic.

egeerts, 2015-04-03:
I can see the error, but as I mention, before yesterday this Designator resolved into a location. Therefore I guess that the update in the designator causes this problem.

closed 2015-04-07 by LoyVanBeek