tue-robotics/tue_robocup #56: Person_recognition never exits when WAIT_FOR_OPERATOR fails

LoyVanBeek, 2015-11-03:
Try with rosrun challenge_person_recognition person_recognition.py mockbot

LFBFerreira, 2015-11-04:
Is there a person in front of the robot... He can't continue the challenge if he indeed can't see anyone to learn a face

LoyVanBeek, 2015-11-04:
If there is no-one in front  of the robot, it should decide to fail and move on after some time.

LFBFerreira, 2015-11-04:
Kind of true, then the rest of the challenge is completly random because it needs to recognize someone. At the time i was confident about detecting a person in front of the robot so i didnt build a workaround for it, but i can change it when i give the challenge an update. Some other states might also crash due to recent changes in ED.

The only challenge that i'm making sure its works, for now, is the test

Rayman, 2015-11-17:
I think that during the challenge, the best action is just to continue waiting. Something else does not make sense.

closed 2016-03-08 by LoyVanBeek