tue-robotics/tue_robocup #62: Challenge person recognition: Drivers to wrong room

reinzor, 2016-03-22:


LoyVanBeek, 2016-03-23:
TODO: Specify room in NavigateToSymbolic

rokusottervanger, 2016-03-23:
When did it drive to the wrong room? When looking for the crowd or to indicate the operator? According to the rulebook, there is not even navigation involved. It should just turn around and look... Why do we even navigate?

rokusottervanger, 2016-03-23:
I believe it happens when the robot needs to navigate to the operator, but there is currently no quick way to fix that. We'd need a way to query in which room the operator is, or to tell the navigation to constrain the goal position by the room an entity is in.

LoyVanBeek, 2016-03-23:
The operator is always in the same room the robot is. Its known beforehand in which room the challenge will take place and it should not move outside that room. 

But you are right that navigation is not strictly required, but taking a good look at someone may need navigation. As long as its clear that the robot found the operator, AFAIK.

reinzor, 2016-04-19:
Fixed during reo2016

closed 2016-04-19 by reinzor