tue-robotics/tue_robocup #114: Local planner not moving, return

reinzor, 2016-04-19:
When the robot does not move anymore, notice and notify the state machine (could be implemented in the state machine or the local planner itself)

@jlunenburg 
@reinzor

jlunenburg, 2016-05-10:
@reinzor Why does the getBlockedPoint function in the local_planner_interface overwrite feedback_.blocked? If computeVelocityCommands (in dwa_planner_ros) does an additional 'isMoving' check and this fails (i.e., the robot has not moved significantly for over 10 seconds) feedback_.blocked will first become true but then false again. 

Alternatively, the isMoving function could be moved to the local_planner_interface or to the state machine (as mentioned above). Maybe we should discuss this next week.

reinzor, 2016-05-11:
I don't know by hard, let's discuss next Tuesday

MatthijsBurgh, 2016-05-18:
Idee van mij en Rokus:
Kunnen we zorgen dat als hij niet beweegt voor een tijdje, dat de alignment meer voorkeur krijgt dan het dichter bij zijn doel komen. Als je dit generieker wilt maken betekent het dat je wilt dat bij lage snelheden alignment meer voorkeur heeft.
Daarnaast kwam ik in de base local planner een preferforward costfunctie tegen. Deze kan misschien nu verminderd worden nu we een laser achter hebben. Hier zit ook een functie in die tegen draaien is.

reinzor, 2016-05-18:
Zit er al in ;)

-Rein

closed 2016-06-18 by jlunenburg