tue-robotics/tue_robocup #121: Follow operator crash

reinzor, 2016-05-03:
[ERROR] [WallTime: 1462306774.117527] InvalidUserCodeError: Could not execute state 'FOLLOW_OPERATOR' of type '<robot_smach_states.navigation.follow_operator.FollowOperator object at 0x7f44e403fb50>': Traceback (most recent call last):
  File "/home/amigo/ros/indigo/system/src/smach/src/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/navigation/follow_operator.py", line 406, in execute
    rospy.sleep(self._period) # Loop at 2Hz
AttributeError: 'FollowOperator' object has no attribute '_period'

reinzor, 2016-05-11:
Is this fixed?

rokusottervanger, 2016-05-11:
yep

closed 2016-05-11 by rokusottervanger