tue-robotics/tue_robocup #137: Initialize state crashes (simulation)

rokusottervanger, 2016-05-25:
[ERROR] [WallTime: 1464181015.296637] Error raised during SMACH container construction: 
Traceback (most recent call last):

  File "/home/rokus/ros/indigo/system/src/challenge_following_and_guiding/src/challenge_following_and_guiding.py", line 86, in setup_statemachine
    smach.StateMachine.add('INITIALIZE', states.Initialize(robot), transitions={   'initialized':'STORE_ROBOCUP_ARENA', 'abort':'aborted'})

  File "/home/rokus/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/utility.py", line 21, in __init__
    start_robocup_recorder(robot.robot_name)

  File "/home/rokus/ros/indigo/system/src/robot_smach_states/src/robot_smach_states/util/robocup_recorder.py", line 22, in start_robocup_recorder
    launch.start()

  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()

  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()

  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)

  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/config.py", line 453, in load_config_default
    raise RLException(e)

RLException: while processing /home/rokus/ros/indigo/system/src/robot_launch_files/launch/record/record_robocup.launch:
while processing /home/rokus/ros/indigo/system/src/sergio_bringup/machines/amigo3.machine:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/rokus/ros/indigo/system/src/sergio_bringup/machines/amigo3.machine'

reinzor, 2016-05-25:
> Assigned #137 <https://github.com/tue-robotics/tue_robocup/issues/137> to
> @reinzor <https://github.com/reinzor>.

Can't solve this due to our current launch structure

-Rein

LoyVanBeek, 2016-05-25:
Apparently, you need to type amig;rosrun challenge_....
Don;t know where this comes from, why its so, but @jlunenburg showed this yesterday.

reinzor, 2016-05-25:
Fixed

closed 2016-05-25 by reinzor