tue-robotics/tue_robocup #142: Laptop bracket on SERGIO

jlunenburg, 2016-06-18:
For the restaurant and follow-and-guide challenges a laptop bracket on the back of SERGIO is desired

hvrooy, 2016-06-18:
Which laptop (size, weight)? 
Will the laptop be opened (and operated) or closed?
Is it a permanent bracket, or must it be (easily) removeable?
Is it needed on the enclosure (torso backplane) later too? (this requires a different or new design)
Is it possible to use a tablet too, or instead?

MatthijsBurgh, 2016-06-18:
15 inch HP laptop
The laptop Will always be open during  challanges. Does not need to be permanent, Just for Leipzig at the moment.
It has to be a Linux machine with Ethernet port. In some challenges, SERGIO will drive away from the arena with a possible chance of losing Wi-Fi connection. Therefore the laptop, which runs the ssh session, need to have a wired connection.

MatthijsBurgh, 2016-06-22:
Done

closed 2016-06-22 by MatthijsBurgh