tue-robotics/tue_robocup #145: Kinetic Migration

Rayman, 2016-08-16:
## Misc stuff
- [ ] xacro --inorder

LoyVanBeek, 2016-08-17:
Is " convert all pip install to apt-get install python-" possible for all our packages?

Rayman, 2016-08-17:
I don't know, we'll have to test this for each package. I found 8 occurences of `pip install`:
https://github.com/tue-robotics/tue-env/search?q="pip+install"

Rayman, 2016-10-25:
@Timple is this branch ready to merge? [amigo_description/kinetic-devel](https://github.com/tue-robotics/amigo_description/compare/kinetic-devel)

Timple, 2016-11-01:
I don't know,

I haven't had an opportunity to test it under indigo.

Feel free to make a backup of amigo.urdf, switch to the kinetic-devel
branch, make and compare the differences.

Regards,

Tim

Timple, 2016-11-08:
Waarom is de package list niet compleet? Ik mis bijvoorbeeld: amigo_hardware

Timple, 2016-11-08:
Why is rosbridge forked?

reinzor, 2016-11-09:
sergio description fixed, I think we can go to Kinetic =D

reinzor, 2016-11-09:
ed rviz plugins also fixed now with qt5

Timple, 2016-11-09:
Our dear Rayman only evaluated the packages of tue-dev.

However amigo does not run on tue-dev. So some more packages (mostly hardware packages) still need to be evaluated!

reinzor, 2016-11-09:
Ah ok, I at least can confirm that the simulator is up and running!

MatthijsBurgh, 2016-12-06:
Saturday, migration of SERGIO?

Timple, 2016-12-08:
Might be worth looking in to! Depends on the priorities I guess

Timple, 2017-02-17:
I propose to close this issue since the kinetic migration was succesfully tested. The open items are nice-to-haves. Not requirements.

closed 2017-02-21 by Rayman