tue-robotics/tue_robocup #146: [EPIC] Object recognition demo

Rayman, 2016-08-30:
## Pre-implementation
- [ ] choose neural network framework
- [ ] architecture

## Implementation
- [ ] Python API
- [ ] node
- [ ] RoboCup data for training
- [ ] annotation tool
- [ ] segmentation sort of stuff

LoyVanBeek, 2016-08-31:
I was experimenting with https://blog.keras.io/building-powerful-image-classification-models-using-very-little-data.html.
In the gdocs TODO list, there's a little investigation into frameworks and Keras (a wrapper for Theano and/or Tensorflow) came out best for us to use.

Rayman, 2016-08-31:
For TensorFlow there is also a (small) tutorial [here](https://www.tensorflow.org/versions/master/how_tos/image_retraining/index.html#using-the-retrained-model). An example can be found here: https://github.com/eldor4do/Tensorflow-Examples/blob/master/retraining-example.py#L82

LoyVanBeek, 2016-09-24:
@rokusottervanger has some annotated data, so we can extract the labeled image patches from that data.
Then we use that to do transfer learning: take a fully trained object detection/recognition net and re-train the top few layers for our application.

Rayman, 2016-09-25:
Where is the data stored? We could try to put it on GitHub LFS: https://git-lfs.github.com/

LoyVanBeek, 2016-09-26:
On his computer at home.

rokusottervanger, 2016-09-27:
Annotated training data, available here for the next 7 days: https://we.tl/1Ag3iq9L6R

Rayman, 2016-09-27:
Unfortunately, git-lfs has some limitations:
- Personal and organization accounts using Git LFS receive 1 GB total for storage
- Personal and organization accounts using Git LFS receive 1 GB a month for bandwidth.

LoyVanBeek, 2016-09-29:
I'm still doubtful that we really need to segment and classify these images manually. 
With the depth images, segmentation should be simpler: where there is a shape edge in both depth and colors, then there is an object boundary. This will not work for all cases, but maybe wit will for the objects in the closet. Then we bootstrap from there.

I think we might also be able to cluster the segmented objects by histogram (or something similar) and apply a label to each cluster's items. 

@reinzor did you have an utility to easily split up the .rgbd-files into color and depth?

reinzor, 2016-09-30:
Lets wait for Rokus, maybe he knows how all the tooling works ...
Hopefully. ​

-Rein

reinzor, 2016-12-10:
https://github.com/tue-robotics/tue_robocup/wiki/Tournament-start-up-guide

closed 2016-12-10 by reinzor