tue-robotics/tue_robocup #148: [EPIC] Robust grasping Demo

MatthijsBurgh, 2016-09-06:
Demo

- 20 round/square items with a 90% success rate.

- no bumping into the table top

- Significant movement of kinect and arm in between of pick-ups

Issues to tackle

-  Chain of inaccuracies
  - Inaccuracies of homing
  - Inaccuracies in the arm
  - Inaccuracies of the localization
  - Inaccuracies of the object localization
    - Inaccuracies of the kinect
      - Inaccuracies in the software
      - Play in the head

Solutions

- [ ] Bench-marking of play in the arm with and without homing in between. Bring gripper to same position multiple times.
- [ ] grasping without moving between looking and grasping to eliminate inaccuracies in localization
- [ ] Bench-marking of object localization, place an object on a table. Enter the pipeline multiple times and save all determined positions.
- [ ] Benchmarking of multiple object on same locations to determine influence of object shape on object localization.
- [ ] Bencmarking of head. Unknown. New, bigger dynamixels.
- [ ] Using gripper marker. According to Rokus, solution is ready for testing.

MatthijsBurgh, 2016-10-26:
I think the issues
#141 #133 
also should be fixed in this demo. This is essential for robust manipulation.

MatthijsBurgh, 2016-12-06:
We are working on the solutions, while no benchmarks are performed. This really needs to be done very soon.

fibonatic, 2017-03-11:
I made a test script for ar markers on the hands of amigo, but there seemed to be quite a big position/rotation error from ar_track_alvar. When testing on amigo, using only rgb images for marker detection, the tf of the markers where placed above and infront of the hands of amigo, instead on its wrists. However the distances between markers seemed correct. It has not been properly tested if including depth information from the kinect (which on initial testing did not seem to improve the detection and positioning) improves this. The branch for this can be found here: https://github.com/tue-robotics/tue_robocup/tree/ar-marker-script/robot_smach_states/scripts

LoyVanBeek, 2017-03-13:
Yes, if I recall correctly, the tape-measure distance from the Kinect lens to the marekr was 47cm while the distance between the TF's was 50 cm. 
Relative distances (between markers) did match within millimeters. 

Also, the robot model of the wrist looks off: the cylinder on the hand does not line up with the corresponding cylinder on the arm.

closed 2017-09-25 by Rayman