tue-robotics/tue_robocup #150: [EPIC] Cleanup robot_skills

Rayman, 2016-12-10:


LoyVanBeek, 2016-12-11:
Wat is to be done with the items in this list? These are the only methods needed?

Rayman, 2016-12-13:
This is a list of everything that *should* be supported. It seems that world_model.py takes the most time

LoyVanBeek, 2016-12-13:
These are the states in use that I can find by ack-grepping in all the challenges, found via
``` bash
roscd challenge_{eegpsr,gpsr,navigation,open,person_recognition,restaurant,rips,speech_recognition,manipulation}
ack-grep --python ".states." --noheading -h >> ~/states.txt
```

The result:

1. - states.AskContinue
2. - states.CancelHead
3. - states.FollowOperator
4. - states.Grab
5. - states.GrabMachineWithoutBase
6. - states.HandoverToHuman
7. - states.HearOptions
8. - states.Initialize
9. - states.LearnPerson
10. - states.LookAtArea
11. - states.LookAtEntity
12. - states.NavigateToObserve
13. - states.NavigateToSymbolic
14. - states.NavigateToWaypoint
15. - states.NavigateWithConstraints
16. - states.OpenDoorByPushing
17. - states.ResetArms
18. - states.ResetArmsSpindle
19. - states.ResetArmsSpindleHead
20. - states.ResetED
21. - states.Say
22. - states.SetInitialPose
23. - states.StartChallengeRobust
24. - states.util.designators.EntityByIdDesignator
25. - states.util.startup
26. - states.WaitForDoorOpen
27. - states.WaitForTrigger
28. - states.WaitTime

LoyVanBeek, 2016-12-13:
The states implemented are:

1. AbortAnalyzer(smach.State)
2. AddNegativeResult(smach.State)
3. AddPositiveResult(smach.State)
4. ArmToJointConfig(smach.State)
5. ArmToJointTrajectory(smach.State)
6. ArmToQueryPoint(smach.State)
7. ArmToUserPose(smach.State)
8. AskContinue(smach.StateMachine)
9. CallFunction(smach.State)
10. CancelHeadGoals(smach.State)
11. CancelHead(smach.State)
12. CheckDoorPassable(smach.State)
13. CheckTime(smach.State)
14. CloseGripperOnHandoverToRobot(smach.State)
15. constraintGenerator(smach.State)
16. Counter(smach.State)
17. EnterArena(smach.StateMachine)
18. Evaluate(smach.State)
19. executePlan(smach.State)
20. FollowOperator(smach.State)
21. ForceDriveToTouchDoor(smach.State)
22. GetEntitysByType(smach.State)
23. getPlan(smach.State)
24. GrabMachine(smach.StateMachine)
25. Grab(smach.StateMachine)
26. GrabWithVisualServoing(smach.State)
27. HandoverFromHuman(smach.StateMachine)
28. HandoverToHuman(smach.StateMachine)
29. HearOptionsExtra(smach.State)
30. HearOptions(smach.State)
31. Hear(smach.State)
32. HearYesNo(smach.State)
33. Initialize(smach.State)
34. InitializeWorldModel(smach.State)
35. Inspect(smach.StateMachine)
36. IteratorState(smach.State)
37. LearnPerson(smach.State)
38. LockDesignator(smach.State)
39. LookAtHand(smach.State)
40. LookAtPersonInFront(smach.State)
41. MarkEntityInRviz(smach.State)
42. Navigate(smach.StateMachine)
43. NavigateTo(smach.StateMachine)
44. OpenDoorByPushing(smach.StateMachine)
45. OpenGripperOnHandoverToHuman(smach.State)
46. PickUp(smach.State)
47. Place(smach.StateMachine)
48. planBlocked(smach.State)
49. PlaySound(smach.State)
50. Point_at_object(smach.State)
51. PointMachine(smach.StateMachine)
52. PrepareEdGrasp(smach.State)
53. PrepareGraspSafe(smach.State)
54. PreparePlace(smach.State)
55. PushPerpendicularToDoor(smach.State)
56. PushSidewaysAndBack(smach.State)
57. Put(smach.State)
58. RandomNav(smach.StateMachine)
59. ResetArmsHead(smach.State)
60. ResetArm(smach.State)
61. ResetArms(smach.State)
62. ResetArmsSpindleHead(smach.State)
63. ResetArmsSpindle(smach.State)
64. ResetED(smach.State)
65. ResetHead(smach.State)
66. ResetHeadSpindle(smach.State)
67. ResetLeftArm(smach.State)
68. ResetOnFailure(smach.StateMachine)
69. ResetRightArm(smach.State)
70. ResetSpindleHeadUp(smach.State)
71. ResetSpindle(smach.State)
72. ResetTorso(smach.State)
73. Say(smach.State)
74. SegmentObjects(smach.State)
75. SelectAction(smach.State)
76. SetGripper(smach.State)
77. SetInitialPose(smach.State)
78. SetSpindle(smach.State)
79. SetTimeMarker(smach.State)
80. StartAnalyzer(smach.State)
81. StartChallengeRobust(smach.StateMachine)
82. State(smach.State)
83. StopAnalyzer(smach.State)
84. Test(smach.StateMachine)
85. TorsoToUserPos(smach.State)
86. Trigger(smach.State)
87. UnlockDesignator(smach.State)
88. WaitCondition(smach.State)
89. WaitForDesignator(smach.State)
90. WaitForPersonDetection(smach.State)
91. WaitForPersonEntity(smach.State)
92. WaitForTrigger(smach.State)
93. WaitTime(smach.State)
94.

reinzor, 2016-12-14:
Maybe an option to automatically wrap all skills in a SmachState so we do not have to maintain those; however, this requires some standardization of the skills.

LoyVanBeek, 2016-12-14:
That would indeed help, but adds more magic. 
Another improvement: 1 generic ResetPart-state, that you pass a list of robot-parts on which .reset will be called.
That requires each robot-part to have a .reset()-method.

LoyVanBeek, 2016-12-14:
Also: what should be the signature of methods that accept e.g. a Point, PointStamped, PoseStamped etc as an argument.
Currently, we have a lot of functions that take x,y,z,rx,ry,rz and a frame_id as input. Makes for handy introspection/completion but it must be unpacked from the robot_smach_states all the time. 
Of course, we can also have multiple methods that have different signatures that wraps each other. 

But which is the way to go?

reinzor, 2016-12-14:
native python types or types we can calculate with; no messages :) . but we
can discuss this.

-Rein

LoyVanBeek, 2016-12-14:
For this, we could use http://www.orocos.org/kdl/usermanual/geometric-primitives, like in https://github.com/tue-robotics/robot_smach_states/blob/f1cd64daeced9371c44fa0e5cf57b1bf2237e72f/src/robot_smach_states/util/geometry_helpers.py

But ED also returns ed.msg.EntityInfo objects, which would then also need to be wrapped in order to be able to calculate with their poses etc.

MatthijsBurgh, 2017-03-28:
@LoyVanBeek Status?

LoyVanBeek, 2017-03-28:
No work yet; I assigned this to myself so I won't forget.

closed 2017-10-23 by Rayman