tue-robotics/tue_robocup #151: Use Entity class instead of plain EntityInfo-message

LoyVanBeek, 2017-01-06:
Setup and use a more general Entity-class. Currently, we use EntityInfo a lot, but much code is duplicated when operating on that data.
This can be reduced when abstracting over EntityInfo with the Entity-class.

class Entity, with attributes:
- pose: KDL.Frame
- type & super_types 
- convex_hull
- volumes
Methods:
- is_a, to check the types and super-types
- distance_to-{2,3}d

LoyVanBeek, 2017-01-06:
This directly makes the switch to KDL datatypes. Maybe we should wait with that and first give Entity a geometry_msgs.Pose first and then convert everything to KDL.Frame. 

Makes the switch easier to grok and test.

LoyVanBeek, 2017-01-07:
Also, a method to convert its pose to a geometry_msgs.Pose wrt. a given frame (defaulting to /map) would be handy in a lot of places as well.

Rayman, 2017-01-07:
For conversion, see all the methods here:
https://github.com/ros/geometry2/blob/indigo-devel/tf2_kdl/src/tf2_kdl/tf2_kdl.py

Looks quite nice, they even have tf2_ros.Stamped versions of KDL vectors.

closed 2017-01-24 by LoyVanBeek