tue-robotics/tue_robocup #161: docopt and graphviz dependency

Timple, 2017-02-11:


LoyVanBeek, 2017-02-12:
robot_smach_states (the already depends on this in it's package.xml and there are target for ros-python-graphviz and ros-python-docopt.

- https://github.com/tue-robotics/tue-env/blob/master/installer/targets/ros-python-graphviz/install.yaml
- https://github.com/tue-robotics/tue-env/blob/master/installer/targets/ros-python-docopt/install.yaml

Both are pip targets but apparently not yet automagically installed. Curious though that those targets include the name "ros-..." in their name

Timple, 2017-02-12:
>    installer/targets/ros-python-graphviz/install.yaml
>    installer/targets/ros-python-docopt/install.yaml

I think that is a flaw in the tue-get package?

Does it automatically look for ros-thingy when thingy is specified in the
package.xml?

MatthijsBurgh, 2017-02-12:
Yes, for all dependencies in package.xml it looks for the target ros-XXX.

see: https://github.com/tue-robotics/tue-env/blob/master/installer/scripts/tue-install-impl#L500

LoyVanBeek, 2017-02-12:
Then I don't know what exactly went wrong.

Timple, 2017-02-17:
Is there maybe a way to differentiate ros-dependencies from system-dependencies in a package.xml?

We could for exampe check first if a ros-dependency exist and otherwise look for a system dependency.

I'm quite sure that is also how ROS itself does that when resolving dependencies.

Rayman, 2017-02-21:
In the package XML you have a `rosdep key`. That resolves to either a system or a ros dependency.

reinzor, 2017-02-21:
@Rayman link?

Rayman, 2017-02-22:
For example [this package.xml](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/package.xml). It has both system and ros dependencies.

LoyVanBeek, 2017-02-22:
Does tue-get look for both run and build depends? python-docopt and python-graphviz are listed only as run_depend: [package.xml](https://github.com/tue-robotics/tue_robocup/blob/master/robot_smach_states/package.xml#L19) of robot_smach_states.

Timple, 2017-02-22:
Yes it does. Mainly used for things in lauch files and such.

MatthijsBurgh, 2017-03-28:
I don't think this is a real issue, because it only happend once. So can I close this one?

closed 2017-04-04 by MatthijsBurgh