tue-robotics/tue_robocup #177: Inspect does not work anymore, because of deeper stuff

MatthijsBurgh, 2017-03-09:
[amigo] > inspect the dinner_table 

    Error message from action server:

        Error while starting action: 'Frame' object has no attribute 'position'

Traceback (most recent call last):
  File "/home/amigo/ros/kinetic/repos/https_/github.com/tue-robotics/action_server.git/src/action_server/server.py", line 60, in add_action_cb
    err = action.start(config, self._robot)
  File "/home/amigo/ros/kinetic/repos/https_/github.com/tue-robotics/action_server.git/src/action_server/actions/action.py", line 17, in start
    return self._start(config, robot)
  File "/home/amigo/ros/kinetic/repos/https_/github.com/tue-robotics/action_server.git/src/action_server/actions/action.py", line 39, in _start
    err = self._init_fsm(config, robot)
  File "/home/amigo/ros/kinetic/repos/https_/github.com/tue-robotics/action_server.git/src/action_server/actions/inspect.py", line 14, in _init_fsm
    (entities, error_msg) = entities_from_description(entity_descr, robot)
  File "/home/amigo/ros/kinetic/repos/https_/github.com/tue-robotics/action_server.git/src/action_server/actions/util/entities_from_description.py", line 41, in entities_from_description
AttributeError: 'Frame' object has no attribute 'position'

LoyVanBeek, 2017-03-09:
Damnit, this is due to the big refactor with the KDL datatypes. I submitted a potential patch

LoyVanBeek, 2017-03-13:
Can I close this?

closed 2017-03-13 by MatthijsBurgh