tue-robotics/tue_robocup #189: Resulting bug in placing, caused by #188

MatthijsBurgh, 2017-03-13:
[ERROR] [1489419911.013969]: InvalidUserCodeError: Could not execute state 'PREPARE_PLACE' of type '<robot_smach_states.manipulation.place.PreparePlace object at 0x7fa5620a4bd0>': Traceback (most recent call last):
  File "/home/amigo/ros/kinetic/system/src/smach/src/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/manipulation/place.py", line 35, in execute
    placement_fs = self._placement_pose_designator.resolve()
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/designators/core.py", line 46, in resolve
    result = self._resolve()
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/designators/ed_designators.py", line 313, in _resolve
    frames_of_interest = self.determine_points_of_interest(place_location)
TypeError: determine_points_of_interest() takes exactly 4 arguments (2 given)

Exception in thread grab:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/amigo/ros/kinetic/system/src/smach/src/smach/state_machine.py", line 359, in execute
    container_outcome = self._update_once()
  File "/home/amigo/ros/kinetic/system/src/smach/src/smach/state_machine.py", line 254, in _update_once
    + traceback.format_exc())
InvalidUserCodeError: Could not execute state 'PREPARE_PLACE' of type '<robot_smach_states.manipulation.place.PreparePlace object at 0x7fa5620a4bd0>': Traceback (most recent call last):
  File "/home/amigo/ros/kinetic/system/src/smach/src/smach/state_machine.py", line 247, in _update_once
    self._remappings[self._current_label]))
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/manipulation/place.py", line 35, in execute
    placement_fs = self._placement_pose_designator.resolve()
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/designators/core.py", line 46, in resolve
    result = self._resolve()
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/designators/ed_designators.py", line 313, in _resolve
    frames_of_interest = self.determine_points_of_interest(place_location)
TypeError: determine_points_of_interest() takes exactly 4 arguments (2 given)

MatthijsBurgh, 2017-03-13:
#188

LoyVanBeek, 2017-03-13:
This looks like the cause of #189, with a weird bug in the call to determine_points_of_interest in https://github.com/tue-robotics/tue_robocup/blob/master/robot_smach_states/src/robot_smach_states/util/designators/ed_designators.py#L313

LoyVanBeek, 2017-03-14:
I think this is also fixed in https://github.com/tue-robotics/tue_robocup/pull/192

MatthijsBurgh, 2017-04-03:
@LoyVanBeek So this one can be closed?

LoyVanBeek, 2017-04-03:
I think so, but I should test this tomorrow

MatthijsBurgh, 2017-04-04:
#192  is merged, so closing this one

closed 2017-04-04 by MatthijsBurgh