tue-robotics/tue_robocup #201: Revise whether we should use catkin build

reinzor, 2017-03-22:
I see all kind of weird linking errors since we are using `catkin build`:

Recently this one:

```
/lib/x86_64-linux-gnu/libc.so.6(+0x354b0)[0x7f21944d14b0]
/home/amigo/ros/kinetic/system/devel/.private/ed_sensor_integration/lib/libed_kinect.so(_ZN3dml12findContoursERKN2cv3MatERKN3geo5Vec2TIiEEiRSt6vectorIS6_SaIS6_EESC_RS1_b+0x128)[0x7f2174653bef]
/home/amigo/ros/kinetic/system/devel/.private/ed_sensor_integration/lib/libed_kinect.so(_ZN3dml16calculateContourERKN2cv3MatERSt6vectorIS4_IN3geo5Vec2TIiEESaIS7_EESaIS9_EE+0x193)[0x7f2174654049]
/home/amigo/ros/kinetic/system/devel/.private/ed_sensor_integration/lib/libed_kinect.so(_ZN3dml9project2DERKN3geo4MeshERSt6vectorIS4_INS0_5Vec2TIdEESaIS6_EESaIS8_EE+0x67d)[0x7f2174654a88]
/home/amigo/ros/kinetic/system/devel/.private/ed_sensor_integration/lib/libed_kinect.so(_ZN6Fitter19GetOrCreateEntity2DERKN5boost10shared_ptrIKN2ed6EntityEEE+0x27e)[0x7f217464a922]
/home/amigo/ros/kinetic/system/devel/.private/ed_sensor_integration/lib/libed_kinect.so(_ZN6Fitter12renderEntityERKN5boost10shared_ptrIKN2ed6EntityEEERKN3geo11Transform3TIdEEiRSt6vectorIdSaIdEERSD_IiSaIiEE+0x1a8)[0x7f217464abbe]
/home/amigo/ros/kinetic/system/devel/.private/ed_sensor_integration/lib/libed_kinect.so(_ZN6Fitter18estimateEntityPoseERK10FitterDataRKN2ed10WorldModelERKNS3_4UUIDERKN3geo11Transform3TIdEERSC_d+0x4e1)[0x7f2174649217]
/home/amigo/ros/kinetic/system/devel/.private/ed_sensor_integration/lib/libed_kinect.so(_ZN7Updater6updateERKN2ed10WorldModelERKN5boost10shared_ptrIKN4rgbd5ImageEEERKN3geo11Transform3TIdEERK13UpdateRequestR12UpdateResult+0x585)[0x7f217466eee5]
/home/amigo/ros/kinetic/system/devel/lib/libed_kinect_plugin.so(_ZN12KinectPlugin9srvUpdateERN21ed_sensor_integration14UpdateRequest_ISaIvEEERNS0_15UpdateResponse_IS2_EE+0x1bc)[0x7f21625d631e]
```

This was due to a change in one of the libs and compiling it with `--this`. For sure this has to do with our set-up but I haven't seen these problems when we were using `catkin_make`. A clean recompile often fixes these problems :/

Timple, 2017-03-28:
Hmmm, that is annoying... 

Can this be prevented by never using `--this`? That would be the same effect as the `catkin_make`.

It is not easy to switch back with all the rtt stuff on amigo. There it is required. Or we need to setup different workspaces and stuff which is not easily supported with a `tue-get install`.

However, feel free to use catkin_make on your own system of course. tue-make will check which one you use and run the proper command.

Also: Would catkin_make be a solution or would it be the masking of a bug? ;-)

My suggestion: stop using `--this` and start using the dev workspace again (should still work with `catkin build`. And then see if the issue dissapears or not.

reinzor, 2017-03-28:
Not sure whether this actually is the problem :/

-Rein

closed 2017-09-25 by Rayman