tue-robotics/tue_robocup #225: Bug if question not answered

jlunenburg, 2017-04-08:
Crashed during runthrough

jlunenburg, 2017-04-08:
Is probably adding skybiometry to launch file

jlunenburg, 2017-04-08:
Traceback (most recent call last):
  File "/home/amigo/ros/kinetic/system/src/challenge_spr/src/challenge_spr.py", line 107, in <module>
    startup(ChallengeSpeechPersonRecognition, challenge_name="challenge_spr")
  File "/home/amigo/ros/kinetic/system/src/robot_smach_states/src/robot_smach_states/util/startup.py", line 91, in startup
    outcome = executioner.execute()
  File "/home/amigo/ros/kinetic/system/src/smach/src/smach/state_machine.py", line 359, in execute
    container_outcome = self._update_once()
  File "/home/amigo/ros/kinetic/system/src/smach/src/smach/state_machine.py", line 266, in _update_once
    self._current_state.get_registered_outcomes()))
smach.exceptions.InvalidTransitionError: Attempted to return outcome 'not_answered' from state 'RIDDLE_GAME_1' of type '<challenge_spr_states.riddle_game.HearQuestion object at 0x7f38bfcb0690>' which only has registered outcomes: ('answered',)

jlunenburg, 2017-04-08:
amigo@amigo2 $ roslaunch challenge_spr.launch 
... logging to /home/amigo/.ros/log/1e26d58e-1c5c-11e7-a82e-00031d0a0d6c/roslaunch-amigo2-4349.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://amigo2:34800/

SUMMARY
========

PARAMETERS
 * /amigo/face_recognition/align_path: ~/openface/models...
 * /amigo/face_recognition/net_path: ~/openface/models...
 * /amigo/face_recognition/save_images: True
 * /amigo/face_recognition/save_images_folder: ~/MEGA/data/robot...
 * /amigo/global_planner/base_global_planner: base_navigation::...
 * /amigo/global_planner/global_costmap/always_send_full_costmap: True
 * /amigo/global_planner/global_costmap/configuration_space/dilation_cell_value: 253
 * /amigo/global_planner/global_costmap/configuration_space/dilation_radius: 0.37
 * /amigo/global_planner/global_costmap/configuration_space/inflation_radius: 0.6
 * /amigo/global_planner/global_costmap/configuration_space/target_cell_value: 254
 * /amigo/global_planner/global_costmap/configuration_space/use_footprint: False
 * /amigo/global_planner/global_costmap/ed_occupancy_grid/map_topic: /amigo/ed/navigat...
 * /amigo/global_planner/global_costmap/ed_occupancy_grid/track_unknown_space: False
 * /amigo/global_planner/global_costmap/global_frame: /map
 * /amigo/global_planner/global_costmap/plugins: [{'type': 'costma...
 * /amigo/global_planner/global_costmap/publish_frequency: 1.0
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/clearing: False
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/data_type: LaserScan
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/inf_is_valid: True
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/marking: True
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/observation_persistence: 0.0
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/obstacle_range: 10.0
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/ray_trace_range: 10.0
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/sensor_frame: /amigo/base_laser
 * /amigo/global_planner/global_costmap/recent_obstacles/base_laser/topic: /amigo/base_laser...
 * /amigo/global_planner/global_costmap/recent_obstacles/observation_sources: base_laser torso_...
 * /amigo/global_planner/global_costmap/recent_obstacles/occupied_to_default_time: 2.0
 * /amigo/global_planner/global_costmap/recent_obstacles/top_kinect/data_type: PointCloud2
 * /amigo/global_planner/global_costmap/recent_obstacles/top_kinect/expected_update_rate: 1.0
 * /amigo/global_planner/global_costmap/recent_obstacles/top_kinect/max_obstacle_height: 0.5
 * /amigo/global_planner/global_costmap/recent_obstacles/top_kinect/min_obstacle_height: 0.0
 * /amigo/global_planner/global_costmap/recent_obstacles/top_kinect/observation_persistence: 0.0
 * /amigo/global_planner/global_costmap/recent_obstacles/top_kinect/obstacle_range: 10
 * /amigo/global_planner/global_costmap/recent_obstacles/top_kinect/sensor_frame: /amigo/top_kinect...
 * /amigo/global_planner/global_costmap/recent_obstacles/top_kinect/topic: /amigo/ed/navigat...
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/clearing: False
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/data_type: LaserScan
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/inf_is_valid: True
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/marking: True
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/observation_persistence: 0.0
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/obstacle_range: 10.0
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/ray_trace_range: 10.0
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/sensor_frame: /amigo/torso_laser
 * /amigo/global_planner/global_costmap/recent_obstacles/torso_laser/topic: /amigo/torso_lase...
 * /amigo/global_planner/global_costmap/resolution: 0.05
 * /amigo/global_planner/global_costmap/robot_base_frame: /amigo/base_link
 * /amigo/global_planner/global_costmap/rolling_window: False
 * /amigo/global_planner/global_costmap/track_unknown_space: False
 * /amigo/global_planner/global_costmap/update_frequency: 1.0
 * /amigo/local_planner/DWAPlannerROS/acc_lim_th: 0.5
 * /amigo/local_planner/DWAPlannerROS/acc_lim_x: 0.35
 * /amigo/local_planner/DWAPlannerROS/acc_lim_y: 0.35
 * /amigo/local_planner/DWAPlannerROS/acc_limit_trans: 0.35
 * /amigo/local_planner/DWAPlannerROS/align_align_scale: 1.0
 * /amigo/local_planner/DWAPlannerROS/align_cmd_nth: 0.0
 * /amigo/local_planner/DWAPlannerROS/align_cmd_nx: 0.0
 * /amigo/local_planner/DWAPlannerROS/align_cmd_ny: 0.0
 * /amigo/local_planner/DWAPlannerROS/align_cmd_pth: 0.0
[... 227 lines cut ...]
[INFO] [1491664695.170617]: 'Sorry, I do not understand your question'
[ERROR] [1491664698.139713]: InvalidTransitionError: Attempted to return outcome 'not_answered' from state 'RIDDLE_GAME_1' of type '<challenge_spr_states.riddle_game.HearQuestion object at 0x7f2840179850>' which only has registered outcomes: ('answered',)
[ERROR] [1491664698.140206]: Robot exited with <class 'smach.exceptions.InvalidTransitionError'>,Attempted to return outcome 'not_answered' from state 'RIDDLE_GAME_1' of type '<challenge_spr_states.riddle_game.HearQuestion object  [... cut ...]
smach.exceptions.InvalidTransitionError: Attempted to return outcome 'not_answered' from state 'RIDDLE_GAME_1' of type '<challenge_spr_states.riddle_game.HearQuestion object at 0x7f2840179850>' which only has registered outcomes: ('answered',)
^C[state_machine-7] killing on exit
[amigo/object_recognition-6] killing on exit
[amigo/global_planner-2] killing on exit
[amigo/face_recognition-5] killing on exit
[amigo/ed-3] killing on exit
[amigo/local_planner-1] killing on exit
^C
^Z
[1]+  Stopped                 roslaunch challenge_spr.launch

MatthijsBurgh, 2017-04-09:
This is cause by missing a possible outcome in https://github.com/tue-robotics/tue_robocup/blob/feature/challenge-spr/challenge_spr/src/challenge_spr_states/riddle_game.py#L54.

This outcome needs to be added in the main challenge, and in the test state machine

closed 2017-05-03 by SamAlexandrov