tue-robotics/tue_robocup #255: TODO RoboCup Japan

Rayman, 2017-05-08:
- improve (Travis) CI
- dedicated vision node
- what to do with the KDL refactor
- improve body part standardization

Timple, 2017-05-10:
Get RGBD out?

LoyVanBeek, 2017-05-10:
The KDL refactor: It is not yet as pretty as I want it to be, but making breaking changes now will definitely suck. 
The most common issue I've heard is that Entities do not have a FrameStamped. I can add this and leave the current Entity.pose (a geometry_msgs.Pose) and Entity._pose (a kdl.Frame) is place and add Entity.frame_stamped. 

After RoboCup Japan, I'll add deprecation markers to ye olde stuff and convert some more stuff, reducing lines of code.

Rayman, 2017-05-10:
Are people using `entity._pose`? That would be scary

LoyVanBeek, 2017-05-10:
:-) It is being used in a couple of places IIRC. This exposed the 'raw' kdl.Frame, while .pose is a property that converts. _pose to a geometry_msgs.msg.Pose to not break everything that depends on it.

LoyVanBeek, 2017-05-10:
- [ ] Do the Setup up a Table and Tidy Up-task.

MaxBaeten, 2017-05-20:
Hardware:
- [ ] fix spindle sensor problem. (at least fix the resetter to check the homing status)

Rayman, 2017-05-20:
- rips: :+1: 
- storing groceries: 55 points (could probably have more points if the kinetic worked)

rokusottervanger, 2017-10-07:
This issue should be closed IMO. Open checkboxes that are still relevant could be moved to new separate issues. @MatthijsBurgh

Rayman, 2017-10-09:
For all items there are issues now (or we're not going to to them). Except these two:

- Is set the table working?
- spr questions related to posture and color

MatthijsBurgh, 2017-10-09:
@SamAlexandrov are questions related to posture and color implemented? Otherwise please create an issue.

MatthijsBurgh, 2017-10-09:
I think we have a working 'Set the table'. Although I think it could be extended.

Rayman, 2017-10-17:
Moved to #437

closed 2017-10-17 by Rayman