tue-robotics/tue_robocup #260: Project yaw angle of a 3d rotation

rokusottervanger, 2017-05-18:
If there are also nonzero roll and pitch angles, this will generally not return the correct projected yaw angle as the other angles may also influence this.
I've so far seen this in 
https://github.com/tue-robotics/tue_robocup/blob/master/robot_smach_states/src/robot_smach_states/navigation/navigate_to_grasp.py#L62 
and 
https://github.com/tue-robotics/tue_robocup/blob/master/robot_smach_states/src/robot_smach_states/navigation/navigate_to_waypoint.py#L42

MatthijsBurgh, 2017-07-16:
What is the status?

Rayman, 2017-11-27:
@rokusottervanger, what do you want to do about this?

MatthijsBurgh, 2017-11-27:
@rokusottervanger and I had a small discussion 1 a 2 weeks ago. GetRPY instead of GetEulerZYX, should solve the problem.

rokusottervanger, 2017-11-28:
You can fix this if you like, @MatthijsBurgh

MatthijsBurgh, 2018-10-25:
GetRPY is not the solution as GetEulerZYX returns GetRPY: http://docs.ros.org/jade/api/orocos_kdl/html/frames_8hpp_source.html#l00493

closed 2021-01-27 by PetervDooren