tue-robotics/tue_robocup #268: Robot places (with wrong hand) an item instead of dropping bag

LoyVanBeek, 2017-05-20:
In the Help-me-carry, the robot must carry a bag. Current implementation assumes it's a rigid object that can be placed on a table. 

Some problems: 

- [ ] Placement happens with wrong arm
- [ ] Placement is on a table while dropping on the ground is also OK

Rayman, 2017-05-20:
The right arm of the robot was not available. A run of the challenge resulted in the following crash:
```
[INFO] [1495294093.163933]: State machine starting in initial state 'PREPARE_PLACE' with userdata: 
	[]
[ERROR] [1495294093.470962054]: No connectivity to specified constraint found, sorry :(
[ERROR] [1495294093.492654293]: No connectivity to specified constraint found, sorry :(
[ERROR] [1495294093.516951786]: No connectivity to specified constraint found, sorry :(
[ERROR] [1495294093.542869038]: No connectivity to specified constraint found, sorry :(
[ERROR] [1495294093.561485882]: No connectivity to specified constraint found, sorry :(
[ERROR] [1495294093.580561223]: No connectivity to specified constraint found, sorry :(
[INFO] [1495294093.707852]: LockingDesignator 'place_position' resolved to pos:(x=3.18, y=4.37, z=0.77), , but is *not locked* to it
[INFO] [1495294093.733513]: Robot has 2 arms
[INFO] [1495294093.754611]: Found 1 available arms: ['right']
[INFO] [1495294093.754774]: Found 0 operational arms: []
[ERROR] [1495294093.755042]: ArmDesignator 

Designator(
  all_arms = OrderedDict([('right', <robot_skills.arms.Arm object at 0x7f6856686a50>), ('left', <robot_skills.arms.Arm object at 0x7f6856662850>)])
  available = <bound method ArmHoldingEntityDesignator.available of Designator(resolve_type=<class 'robot_skills.arms.Arm'>, name=arm_with_item_designator)>
  entity_designator = 

Designator(
  ed = <robot_skills.world_model_ed.ED object at 0x7f685663c550>
  id_ = coke
  lockable = <bound method EntityByIdDesignator.lockable of Designator(resolve_type=<class 'robot_skills.util.entity.Entity'>, name=current_item)>
  name = current_item
  resolve_type = <class 'robot_skills.util.entity.Entity'>)

  lockable = <bound method ArmHoldingEntityDesignator.lockable of Designator(resolve_type=<class 'robot_skills.arms.Arm'>, name=arm_with_item_designator)>
  name = arm_with_item_designator
  preferred_arm = <robot_skills.arms.Arm object at 0x7f6856686a50>
  resolve_type = <class 'robot_skills.arms.Arm'>)

 could not resolve to an arm
[ERROR] [1495294093.755170]: Could not resolve arm
[INFO] [1495294093.755281]: State machine terminating 'PREPARE_PLACE':'failed':'failed'
[INFO] [1495294093.755400]: State machine terminating 'PUTDOWN_ITEM':'failed':'Aborted'
Final outcome: Aborted
[INFO] [1495294093.755653]: Torso cancelling all goals on close
[INFO] [1495294093.755817]: left arm cancelling all goals on all arm-related ACs on close
[INFO] [1495294093.756135]: right arm cancelling all goals on all arm-related ACs on close
[amigo1-0] killing on exit
[amigo1-0]: [amigo/robcup_recorder-10] killing on exit
[amigo1-0]: [amigo/record_throttle_odom-8] killing on exit
[amigo1-0]: [amigo/record_relay_base_plan-6] killing on exit
[amigo1-0]: [amigo/record_throttle_gmapping_map-5] killing on exit
[amigo1-0]: [amigo/hmi/hmi_logger-1] killing on exit
[amigo1-0]: [amigo/record_throttle_tf-3] killing on exit
================================================================================REQUIRED process [state_machine-7] has died!
process has finished cleanly
log file: /home/amigo/.ros/log/f454e726-3d6e-11e7-b7d6-00031d0a0d6c/state_machine-7*.log
Initiating shutdown!
================================================================================
```

LoyVanBeek, 2017-05-23:
This is being worked on in https://github.com/tue-robotics/tue_robocup/issues/295
Solves the first of the two issues

closed 2017-06-06 by LoyVanBeek