tue-robotics/tue_robocup #283: Check status of robot Bodyparts at challenge start

LoyVanBeek, 2017-05-23:
For each bodypart, report when they are idle.

Do not block on this, just report and continue, as suggested by @MaxBaeten 
After this report, the operator must take action.

MatthijsBurgh, 2017-06-24:
PR is merged, so done? @LoyVanBeek

LoyVanBeek, 2017-06-25:
The merged R is about the e-stop, not all other body parts. So not yet done but n=does not have high prio either.

LoyVanBeek, 2018-01-13:
With the SSL on a separate board that's hard to see the status of, this is bound to go wrong sometime...

MatthijsBurgh, 2018-01-13:
Checking on publishing can be tricky, because it only publishes, when a new detection is received.

LoyVanBeek, 2018-01-13:
I know. I just check if there is at least one publisher on the topic, not counting messages

closed 2018-05-08 by jlunenburg