tue-robotics/tue_robocup #320: Speech recognition crashes

LoyVanBeek, 2017-06-13:
```
Jun 13 20:56:27 amigo1 sudo[953]: [ERROR] [1497380187.737737]: _determine_answer raised an exception: Traceback (most recent call last):
Jun 13 20:56:27 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/hmi/src/hmi/abstract_server.py", line 33, in _execute_cb
Jun 13 20:56:27 amigo1 sudo[953]:     is_preempt_requested=self._is_preempt_requested)
Jun 13 20:56:27 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/hmi_server_dragonfly_client.py", line 19, in _determine_answer
Jun 13 20:56:27 amigo1 sudo[953]:     result = c.recognize(grammar, target, is_preempt_requested)
Jun 13 20:56:27 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/dragonfly_client.py", line 35, in recognize
Jun 13 20:56:27 amigo1 sudo[953]:     sentence = conn.recv()
Jun 13 20:56:27 amigo1 sudo[953]: EOFError
Jun 13 20:56:42 amigo1 sudo[953]: [WARN] [1497380202.642085]: no result received
Jun 13 20:56:44 amigo1 sudo[953]: [ERROR] [1497380204.673931]: _determine_answer raised an exception: Traceback (most recent call last):
Jun 13 20:56:44 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/hmi/src/hmi/abstract_server.py", line 33, in _execute_cb
Jun 13 20:56:44 amigo1 sudo[953]:     is_preempt_requested=self._is_preempt_requested)
Jun 13 20:56:44 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/hmi_server_dragonfly_client.py", line 19, in _determine_answer
Jun 13 20:56:44 amigo1 sudo[953]:     result = c.recognize(grammar, target, is_preempt_requested)
Jun 13 20:56:44 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/dragonfly_client.py", line 35, in recognize
Jun 13 20:56:44 amigo1 sudo[953]:     sentence = conn.recv()
Jun 13 20:56:44 amigo1 sudo[953]: EOFError
Jun 13 20:56:59 amigo1 sudo[953]: [WARN] [1497380219.603365]: no result received
Jun 13 20:57:01 amigo1 sudo[953]: [ERROR] [1497380221.606744]: _determine_answer raised an exception: Traceback (most recent call last):
Jun 13 20:57:01 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/hmi/src/hmi/abstract_server.py", line 33, in _execute_cb
Jun 13 20:57:01 amigo1 sudo[953]:     is_preempt_requested=self._is_preempt_requested)
Jun 13 20:57:01 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/hmi_server_dragonfly_client.py", line 19, in _determine_answer
Jun 13 20:57:01 amigo1 sudo[953]:     result = c.recognize(grammar, target, is_preempt_requested)
Jun 13 20:57:01 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/dragonfly_client.py", line 35, in recognize
Jun 13 20:57:01 amigo1 sudo[953]:     sentence = conn.recv()
Jun 13 20:57:01 amigo1 sudo[953]: EOFError
Jun 13 20:57:15 amigo1 sudo[953]: [ERROR] [1497380235.036546]: Error processing request: 'NoneType' object has no attribute 'close'
Jun 13 20:57:15 amigo1 sudo[953]: ['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/amigo/ros/kinetic/system/src/text_to_speech/scripts/text_to_speech_node.py", line 173, in speak_srv\n    self.do_tts(req)\n', '  File "/home/amigo/ros/kinetic/system/src/text_to_speech/scripts/text_to_speech_node.py", line 69, in do_tts\n    self.do_tts_philips(req)\n', '  File "/home/amigo/ros/kinetic/system/src/text_to_speech/scripts/text_to_speech_node.py", line 114, in do_tts_philips\n    resp = self.client_play(play_req)\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__\n    return self.call(*args, **kwds)\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 528, in call\n    transport.close()\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 838, in close\n    self.socket.close()\n', "AttributeError: 'NoneType' object has no attribute 'close'\n"]
Jun 13 20:57:15 amigo1 sudo[953]: [ERROR] [1497380235.133925]: _determine_answer raised an exception: Traceback (most recent call last):
Jun 13 20:57:15 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/hmi/src/hmi/abstract_server.py", line 33, in _execute_cb
Jun 13 20:57:15 amigo1 sudo[953]:     is_preempt_requested=self._is_preempt_requested)
Jun 13 20:57:15 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/hmi_server_dragonfly_client.py", line 19, in _determine_answer
Jun 13 20:57:15 amigo1 sudo[953]:     result = c.recognize(grammar, target, is_preempt_requested)
Jun 13 20:57:15 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/dragonfly_client.py", line 35, in recognize
Jun 13 20:57:15 amigo1 sudo[953]:     sentence = conn.recv()
Jun 13 20:57:15 amigo1 sudo[953]: EOFError
Jun 13 20:57:30 amigo1 sudo[953]: [WARN] [1497380250.054022]: no result received
Jun 13 20:57:32 amigo1 sudo[953]: [ERROR] [1497380252.120962]: _determine_answer raised an exception: Traceback (most recent call last):
Jun 13 20:57:32 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/hmi/src/hmi/abstract_server.py", line 33, in _execute_cb
Jun 13 20:57:32 amigo1 sudo[953]:     is_preempt_requested=self._is_preempt_requested)
Jun 13 20:57:32 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/hmi_server_dragonfly_client.py", line 19, in _determine_answer
Jun 13 20:57:32 amigo1 sudo[953]:     result = c.recognize(grammar, target, is_preempt_requested)
Jun 13 20:57:32 amigo1 sudo[953]:   File "/home/amigo/ros/kinetic/system/src/dragonfly_speech_recognition/src/dragonfly_speech_recognition/dragonfly_client.py", line 35, in recognize
Jun 13 20:57:32 amigo1 sudo[953]:     sentence = conn.recv()
Jun 13 20:57:32 amigo1 sudo[953]: EOFError
Jun 13 20:57:35 amigo1 sudo[953]: [WARN] [1497380255.830258]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
Jun 13 21:00:40 amigo1 sudo[953]: [WARN] [1497380440.619382]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
Jun 13 21:02:34 amigo1 sudo[953]: 9939.633 [ Warning][ScriptingService] Stopped driving because the reference-flow was interrupted.
```

LoyVanBeek, 2017-06-13:
I have no logs for the windows side yet, unfortunately.

Rayman, 2017-10-03:
Without logs, we can't fix this

closed 2017-10-03 by Rayman