tue-robotics/tue_robocup #322: Knowledge objects and classification labels not in sync

rokusottervanger, 2017-06-13:
In 34ad25e811209a9b1cfcba804ff66283bf115513, a number of objects have been removed from the common knowledge for robotics_testlabs (among which the legendary bifrutas). Currently, the following objects are in the labels file in the MEGA: 
- beer
- bifrutas
- coke
- deodorant
- fanta
- ice tea
- mentos
- sponge
- tea
- water
A large part of which is no longer in the knowledge now. I think that the knowledge should at least include the items that can be classified, so the knowledge should be updated. But also we could update the classifcation model so that more objects are included.

MatthijsBurgh, 2017-06-15:
We switched to the object set of the GO, but this one is also complete anymore, so maybe merge the two sets and train the network again.

jlunenburg, 2017-06-27:
@Rayman will fix this later today.

Note: there is a testscript (analyse-annotations) in the package test_tools. Although it does not yet test the knowledge with the labels in output.txt, it still gives an impression if things are correct

Rayman, 2017-11-27:
The script in `test_tools` can be used for detecting this now

closed 2017-11-27 by Rayman

MatthijsBurgh, 2017-11-27:
fixed by a9505acb54411b088a9c30e58835441ff9284095 and d44287ed101b538b7c828a8ffd80698fc77bb639