tue-robotics/tue_robocup #329: GPSR test commands

SamAlexandrov, 2017-06-27:
Category I:
   - [ ] Tell me the name of the person in the kitchen
   - [ ] Get the noodles from the dining table and put it on the cupboard
   - [ ] 
   - [ ] 
   - [ ] 
Category II:
   - [ ] 
   - [ ] 
   - [ ] 
Category III:
   - [ ] 
   - [ ] 
   - [ ]

rokusottervanger, 2017-07-18:
Testing can be done as follows:
 - Start the simulator: 
```
amigo-start
```
 - Start the free mode, the --screen option lets the action server print output to the screen so that you can clearly see what's going on in case of an error:
```
amigo-free-mode --screen
```
 - Start the natural language console, the --debug makes sure that the semantics parsed from the sentence are printed to the screen:
```
robot-console --debug
```
 - The natural language console loads its grammar from the GPSR knowledge, so you can use this to pass tasks to the robot. Also you get tab completion (after the first word, at least, that can still be fixed, if you think it's important). If it doesn't complete (after the first word), it's not in the grammar:
```
[amigo] > get [tab]
[amigo] > get the [tab]
[amigo] > get the noodles [tab]
[amigo] > get the noodles from
...
```
until you get something like 
```
[amigo] > get the noodles from the cabinet and put them on the dinner_table
```
 - Check if the robot does what it is supposed to do. If it doesn't, fix it if you know what you're doing, otherwise ask me. If it does, throw a party, but not for too long, because there are probably other tasks that need to be fixed ;)

LoyVanBeek, 2017-07-18:
Note: you can also pass a sentence directly to the robot_console:
```bash
robot-console --debug "get the noodles from the cabinet and put them on the dinner_table"
```

rokusottervanger, 2017-07-18:
Although that doesn't give you tab completion ;)

Rayman, 2017-11-21:
Command generator might change in the rulebook, let's close for now

closed 2017-11-21 by Rayman