tue-robotics/tue_robocup #330: TEST: SPR commands

SamAlexandrov, 2017-06-27:
> _Please tick only if it's tested on the robot and it's successfully done! The italic text is just a remark. 
All commands are applicable directly for Japan, so they could be tested even during the setup days!
For our environment please change some things like "manju" or "sink"._

Category Predefined:

   - [ ] 
   - [ ] 
   - [ ] 
   - [ ] 
   - [ ] 

Category Arena:

   - [ ] 
   - [ ] 
   - [ ] 
   - [ ] 
   - [ ] 

Category Crowd:

   - [ ] 
   - [ ] 
   - [ ] 
   - [ ] 
   - [ ] 

Category Object:

   - [ ] 
   - [ ] 
   - [ ] 
   - [ ] 
   - [ ]

MatthijsBurgh, 2017-10-03:
[INFO] [1507060170.330877]: Robot heard 'how many bar are in the kitchen' {u'actions': [{u'action': u'count_placement', u'location': u'kitchen', u'entity': u'bar'}]}
[INFO] [1507060170.331095]: Question was: 'how many bar are in the kitchen'?
[INFO] [1507060170.331417]: 'The answer is There are no bar in the livingroom'

MatthijsBurgh, 2017-10-03:
[INFO] [1507060692.134382]: State machine transitioning 'REQUEST_OPERATOR':'spoken'-->'WAIT_FOR_OPERATOR'
WaitForDesignator: waiting 0/3
[ERROR] [1507060692.144836]: No entities found in 

Designator(
  center_point = [           0,           0,           0] @ /map
  center_point_designator = None
  criteriafuncs = []
  debug = False
  ed = <robot_skills.world_model_ed.ED object at 0x7f7eb7158b10>
  id = 
  id_designator = None
  lockable = <bound method EdEntityDesignator.lockable of Designator(resolve_type=<class 'robot_skills.util.entity.Entity'>, name=None)>
  name = None
  radius = 0
  resolve_type = <class 'robot_skills.util.entity.Entity'>
  robot = <robot_skills.amigo.Amigo object at 0x7f7eb7970590>
  type = human
  type_designator = None
  weight_function = <function <lambda> at 0x7f7eb6fd9d70>)

WaitForDesignator: waiting 1/3
[ERROR] [1507060693.158963]: No entities found in 

Designator(
  center_point = [           0,           0,           0] @ /map
  center_point_designator = None
  criteriafuncs = []
  debug = False
  ed = <robot_skills.world_model_ed.ED object at 0x7f7eb7158b10>
  id = 
  id_designator = None
  lockable = <bound method EdEntityDesignator.lockable of Designator(resolve_type=<class 'robot_skills.util.entity.Entity'>, name=None)>
  name = None
  radius = 0
  resolve_type = <class 'robot_skills.util.entity.Entity'>
  robot = <robot_skills.amigo.Amigo object at 0x7f7eb7970590>
  type = human
  type_designator = None
  weight_function = <function <lambda> at 0x7f7eb6fd9d70>)

WaitForDesignator: waiting 2/3
[ERROR] [1507060694.175811]: No entities found in 

Designator(
  center_point = [           0,           0,           0] @ /map
  center_point_designator = None
  criteriafuncs = []
  debug = False
  ed = <robot_skills.world_model_ed.ED object at 0x7f7eb7158b10>
  id = 
  id_designator = None
  lockable = <bound method EdEntityDesignator.lockable of Designator(resolve_type=<class 'robot_skills.util.entity.Entity'>, name=None)>
  name = None
  radius = 0
  resolve_type = <class 'robot_skills.util.entity.Entity'>
  robot = <robot_skills.amigo.Amigo object at 0x7f7eb7970590>
  type = human
  type_designator = None
  weight_function = <function <lambda> at 0x7f7eb6fd9d70>)

[INFO] [1507060695.176554]: State machine transitioning 'WAIT_FOR_OPERATOR':'failed'-->'RIDDLE_GAME'

MatthijsBurgh, 2017-12-08:
@SamAlexandrov If this need to be tested, test it tomorrow. Otherwise close it.

MaxBaeten, 2018-01-09:
Since this is not really an issue. let's move this elswhere

closed 2018-01-09 by MaxBaeten