tue-robotics/tue_robocup #333: RGBD

MatthijsBurgh, 2017-06-27:
Despite the bad stability of the kinect driver, we should not spent too much time on this one. I think next Tuesday is really the deadline for this.

If removing all the rgbd is not working. I propose to use the freenect driver and a ros_to_rgbd encoder.
This way we keep all the functionality and the driver should be way more stable.

reinzor, 2017-06-28:
I agree that we shouldn't spend more time on this. By up until now, the main problem is the hardware that uses connection. So I think we should keep the software as is.

LoyVanBeek, 2017-06-28:
If the HW is shit then software of course won't fix it. Matthijs, what's the status of the search for a replacement Kinec? Or is it just the cables, power supply?

MatthijsBurgh, 2017-06-28:
I don't know the status of that. Both Rein and Max, worked on that yesterday

LoyVanBeek, 2017-07-04:
Today, when I came in the Kinect was powered externally and the USB was connected with a separate cable plugged into the front of Amigo. 

This worked like a charm, no problems for at least half an hour. 

Then I wanted to move Amigo around and thus plugged in the Kinect into the original stuff on Amigo itself. Then it crashes all the time.

Rayman, 2017-07-04:
This is the log:
```
process[amigo/top_kinect/driver-10]: started with pid [11583]
[amigo/top_kinect/driver-10] process has died [pid 11583, exit code 1, cmd /home/amigo/ros/kinetic/system/devel/lib/kinect_driver/kinect_driver __name:=driver __log:=/home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10.log].
log file: /home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10*.log
[amigo/top_kinect/driver-10] restarting process
process[amigo/top_kinect/driver-10]: started with pid [11612]
[amigo/top_kinect/driver-10] process has died [pid 11612, exit code 1, cmd /home/amigo/ros/kinetic/system/devel/lib/kinect_driver/kinect_driver __name:=driver __log:=/home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10.log].
log file: /home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10*.log
[amigo/top_kinect/driver-10] restarting process
process[amigo/top_kinect/driver-10]: started with pid [11642]
[amigo/top_kinect/driver-10] process has died [pid 11642, exit code 1, cmd /home/amigo/ros/kinetic/system/devel/lib/kinect_driver/kinect_driver __name:=driver __log:=/home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10.log].
log file: /home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10*.log
[amigo/top_kinect/driver-10] restarting process
process[amigo/top_kinect/driver-10]: started with pid [11672]
[amigo/top_kinect/driver-10] process has died [pid 11672, exit code 1, cmd /home/amigo/ros/kinetic/system/devel/lib/kinect_driver/kinect_driver __name:=driver __log:=/home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10.log].
log file: /home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10*.log
[amigo/top_kinect/driver-10] restarting process
process[amigo/top_kinect/driver-10]: started with pid [11701]
[amigo/top_kinect/driver-10] process has died [pid 11701, exit code 1, cmd /home/amigo/ros/kinetic/system/devel/lib/kinect_driver/kinect_driver __name:=driver __log:=/home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10.log].
log file: /home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10*.log
[amigo/top_kinect/driver-10] restarting process
process[amigo/top_kinect/driver-10]: started with pid [11733]
[amigo/top_kinect/driver-10] process has died [pid 11733, exit code 1, cmd /home/amigo/ros/kinetic/system/devel/lib/kinect_driver/kinect_driver __name:=driver __log:=/home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10.log].
log file: /home/amigo/.ros/log/b1d3d58e-60e5-11e7-8877-00031d0a0d6c/amigo-top_kinect-driver-10*.log
```

LoyVanBeek, 2017-07-04:
To try: plug in kinect USB via separate extender cable plugged into front ports of amigo1. 
No crashes on first launch while this was a problem just before with the built-in cable.

LoyVanBeek, 2017-07-15:
@hvrooy this is solved now right, with grounding the torso to the base?

hvrooy, 2017-07-16:
Problem is really fixed. Grounding from torso to base was not connected properly. Now it is.

closed 2017-07-16 by hvrooy