tue-robotics/tue_robocup #342: Improve on diagnostics

Timple, 2017-07-10:
[ ] System load on all PCs
[ ] Timing differences between PCs
[ ] Output of kernel loggings (we do not expect any after boot is completed)

MatthijsBurgh, 2021-01-27:
I like this one. @Timple did you implement this somewhere else, which we can use?

Timple, 2021-01-27:
Yes! For system load we use [rosmon](http://wiki.ros.org/rosmon) to launch our applications. This gives memory and cpu usage per rosnode.

We have our own chrony diagnostics for the second point. Let me see what I can do there.

Also for systemd diagnostics (services stopped/failed) we have a diagnostic node.

Timple, 2021-01-28:
Example systemd service entry for rosmon:

```
[Unit]
Description=Amigo hardware launch
After=roscore.service
# Restart if roscore restarts
BindsTo=roscore.service

[Install]
WantedBy=amigo.target

[Service]
User=amigo
# roslaunch is always happy, so not relying on exit code
Restart=always
RestartSec=5
ExecStart=/bin/bash -c 'source /home/amigo/.tue/setup.bash && mon launch --disable-ui --flush-stdout --name=hardware amigo hardware.launch'
TasksMax=infinity
KillSignal=SIGINT
TimeoutStopSec=5
```