tue-robotics/tue_robocup #348: KDL doing magic stuff

MatthijsBurgh, 2017-07-18:
Not working in amigo-console and in rosrun
```
robot.base.get_location().frame.p
[           0,     4.14262,           0]

location = robot.base.get_location()
location.frame.p
[     1.36757,     4.14262,           0]

frame = robot.base.get_location().frame
frame.p
[     1.36757,     4.14262,           0]

p = robot.base.get_location().frame.p
p
[           0,     4.14262,           0]

robot.base.get_location().extractVectorStamped()
[           0,     4.14262,           0] @ /map

VS = robot.base.get_location().extractVectorStamped()
VS
[           0,     4.14262,           0] @ /map

location = robot.base.get_location()
location.extractVectorStamped()
[     1.36757,     4.14262,           0] @ /map

VS = location.extractVectorStamped()
VS
[     1.36757,     4.14262,           0] @ /map

fs = kdl_conversions.FrameStamped(kdl.Frame(kdl.Rotation.Quaternion(1, 0, 0, 0), kdl.Vector(1, 2, 3)), "/map")
fs.extractVectorStamped()
[           1,           2,           3] @ /map
```

LoyVanBeek, 2017-07-18:
What's funny is that only the x value is acting weird. 

I'm suspecting some pointed-at values getting lost or a simple conversion issue

MatthijsBurgh, 2017-07-18:
extractVectorStamped() can be fixed by using ```return deepcopy(XXX)```

MatthijsBurgh, 2017-07-18:
```
import PyKDL as kdl
rot = kdl.Rotation.Quaternion(1, 0, 0, 0)
trans = kdl.Vector(1, 2, 3)
frame=kdl.Frame(rot, trans)
print "\nframe.p"
print frame.p
print "\nkdl.Frame(frame)"
print kdl.Frame(frame)
print "\nkdl.Frame(frame).p"
print kdl.Frame(frame).p
```

This shows the real fault. No code from us at all. So It has to be a bug in PyKDL

MatthijsBurgh, 2017-07-18:
Issue is already known:
orocos/orocos_kinematics_dynamics#41

MatthijsBurgh, 2018-10-17:
Also solved by building PyKDL from source.

closed 2018-10-17 by MatthijsBurgh