tue-robotics/tue_robocup #351: Is the service /person_detection/detect_persons still used?

MaxBaeten, 2017-07-24:
Is the service /person_detection/detect_persons still used?

amigo@ompc-robotics $ ack-grep -R "person_detection" . 
tue_robocup.git/challenge_final/scripts/detect_persons_caller.py
20:print "Waiting for amigo/person_detection/detect_persons"
21:rospy.wait_for_service('/amigo/person_detection/detect_persons')
22:detect_persons = rospy.ServiceProxy('/amigo/person_detection/detect_persons', GetPersons)

tue_robocup.git/robot_skills/src/robot_skills/head.py
91:        self._get_persons_srv = self.create_service_client('/' + robot_name + '/person_detection/detect_persons', GetPersons)
[~/ros/kinetic/repos/https_/github.com/tue-robotics]

MatthijsBurgh, 2017-07-24:
API is broken,  because getPerson srv is removed from image_recognition_msgs. I don't see it directly how to fix it.

MatthijsBurgh, 2017-09-20:
API is changed, fully implemented

closed 2017-09-20 by MatthijsBurgh