tue-robotics/tue_robocup #375: Switch from IPs to local address resolving

Timple, 2017-09-04:
**Proposal** Get rid of all the fixed ip's on dev-pcs etc and switch to local address resolving. 

**Examples**
`ping amigo1.local`
`ssh amigo@amigo2.local`
`export ROS_MASTER_URI=http://amigo1.local:11311`

**Rationale** The custom /etc/hosts files are a hassle to maintain. Also every tournament our subnet changes which makes everything difficult.

**How does it work** If an avahi client and deamon is installed on every PC (simple apt-gettie) then every PC acts as a DNS resolver for only it's own hostname.local. So a ping amigo1.local results in the user PC broadcasting the question who amigo1.local is. ONLY amigo1 responds with the answer: I am here with this IP. Then the action ping starts.
If a computer has multiple IP addresses this still works

**Pros**
- DHCP can be used for most PCs
- Internet access and standard DNS resolving becomes easier because of DHCP
- We don't need to be in control of the router 
- Swapping between cable and wifi on laptops is easier as ever
- Roll-out can be incremental, minimizing risks

**Cons**
- DHCP server required in the network, so a workaround is needed for amigo itself

**Why now**
- Got a call about networking, really needs to be easier!
- Newly gained knowledge about (local) dns resolution ;-)
- Can be part of the 'we need to be more robust' plan

**Todo**

- [ ] Install and configure avahi deamons on every PC
- [ ] Investigate how to do DHCP and a static IP on amigo PCs
- [ ] Configure static IPs on amigo PCs

Rayman, 2017-09-05:
The last point (`export ROS_HOSTNAME=``hostname``.local`) looks to me like a hack so you don't have to change a lot of code.

Timple, 2017-09-05:
You realize we now do: 

``export ROS_IP=`hostname -I | sed 's/.*\(192\.168\.44\.[0-9]\+\).*/\1/'` ``

Talking about hacks :P

But seriously, why does setting the ROS_HOSTNAME look like a hack? You are simply telling the other ROS nodes on other PCs that this is the address where you can be found.

Rayman, 2017-09-06:
Yeah that's true, its way better then what we had before

closed 2018-10-17 by MatthijsBurgh