tue-robotics/tue_robocup #386: Presentation

jlunenburg, 2017-09-12:
So that demo team knows what to tell.

Needs to be delegated/divided

LarsJanssenTUe, 2017-09-21:
[Prestentation_TechUnited_Home.pdf](https://github.com/tue-robotics/tue_robocup/files/1321006/Prestentation_TechUnited_Home.pdf)

Hierboven de presentatie zoals hij tot nu toe is. Graag zou ik van jullie wat input hebben over dingen die toegevoegd of juist geschrapt moeten worden. Andere op en of aanmerkingen zijn ook van harte werlkom!

jlunenburg, 2017-09-21:
Good busy! Is het een idee om je source files (latex neem ik aan?) ook op Github in te checken zoals b.v. TDP en Team Poster. Dat is handig met feedback geven (merge request?).. Evt. kunnen we automatisch laten compilen (zie ook @Home rulebook). 

@MatthijsBurgh mssn is het handig om 1 repo (tue-docs o.i.d.) te maken met dit, team poster en tdp.

LoyVanBeek, 2017-09-25:
Nice work! 

Is het een idee om dit helemaal tot presentatie te verwerken, met slides etc? Ik weet niet of je met LaTeX beamer ook tekst voor alleen de presentator kunt weergeven, als hints?

closed 2020-02-02 by LarsJanssenTUe