tue-robotics/tue_robocup #393: SPR count bug: 2 detections but only 1 male?

Rayman, 2017-09-12:
2 detections but only 1 male?
```
[INFO] [1505249007.178229]: 'You are all looking great today!            Keep looking in my camera!'
[INFO] [1505249008.614142]: getting one image...
[INFO] [1505249008.614302]: self._camera_last_image: None
[INFO] [1505249008.645870]: len(self._camera_last_image): 3507018
[INFO] [1505249008.646051]: got 3507018 bytes of image data
[INFO] [1505249009.664988]: Faces: [RecognizedFace(roi=(773, 284, 127, 150), l2_distances=[]), RecognizedFace(roi=(462, 248, 128, 150), l2_distances=[])]
[INFO] [1505249009.674833]: Calling Skybiometry...
[INFO] [1505249012.937637]: face_properties:
 - gender: 0
age: 27
 - gender: 0
age: 67
[INFO] [1505249012.937981]: 'There are 1 males and 0 females in the crowd'
[INFO] [1505249016.828295]: State machine transitioning 'DETECT_CROWD':'succeeded'-->'REQUEST_OPERATOR'
```

SamAlexandrov, 2017-10-17:
It was tested a couple of times and hasn't occurred anymore.

closed 2017-10-17 by SamAlexandrov

alberth, 2017-10-18:
Elderly people aren't counted as male/female
https://github.com/tue-robotics/tue_robocup/blob/8cba8d170b80605c4f2ee4e2d83d5248ad93179e/challenge_spr/src/challenge_spr_states/detect.py#L132