tue-robotics/tue_robocup #395: Research object recognition

reinzor, 2017-09-19:
[ ] Study TDP and create an overview of the used methods by other teams https://goo.gl/pFMBWk
[ ] Search the web for state of the art detection algorithms
[ ] Try the detect net example from nvidia dusty page (start with digits setup)
[ ] RGBD Sensor selection (also investigate ROS drivers)
[ ] Investigate if it's feasible to train your own detectnet wich such a small dataset, retraining?
[ ] Investigate deployment Jetsons using tensorrt.
[ ] investigate performance of current method.
[ ] What cloud based object recognition services do exist and how do they perform? Custom labels possible?

@alberth @JosjaG @LoyVanBeek

reinzor, 2017-09-19:
Current implementation: https://github.com/tue-robotics/image_recognition

MatthijsBurgh, 2017-09-26:
Might be useful, ROS wrapper of darknet: https://github.com/leggedrobotics/darknet_ros

reinzor, 2017-09-27:
APC vision https://github.com/andyzeng/apc-vision-toolbox#deep-learning-fcn-ros-package

reinzor, 2017-10-03:
Useful for selecting 3D sensor https://roscon.ros.org/2017/presentations/ROSCon%202017%203D%20Vision%20Technology.pdf

LoyVanBeek, 2017-10-16:
From @alberth 

aiRobots_NCKU.pdf
=================
For real-time image processing and object recognition system, we
implemented the affine SIFT scheme [5] and the integration of SURF and
BRISK methods [6].  Our paper [6] proposed a real-time object recognition
systems for recognizing known objects and searching unknown objects for home
service robot. The object recognition system is mainly used in recognizing
known objects, which combined Compute Unified Device Architecture (CUDA),
Speeded Up Robust Features (SURF) detector and Binary Robust Invariant
Scalable Keypoints (BRISK) descriptor for improving the computation speed
and decreasing the consumption on memory. On the other side, the visual
perception system is usually used for searching unknown objects, which
calculated the depth differences and found the contours of objects.

[5]. Y.-H. Wang, H.-E. Cheng, C.-J. Lin, R.-W. Deng, H. Lee, and T.-H. S.
Li, “Realization of Affine SIFT Real-time Image Processing for Home
Service Robot,” in Proc. IEEE ICSSE 2013, IEEE Xplore, DOI:
10.1109/ICSSE.2013.6614648, July 2013, Budapest, Hungary.
[6]. H. Lee, C.-Y. Liu, C.-J. Lin, C.-F. Huang, R.-W. Deng, and T.-H. S. Li,
“Implementation of Real-time Object Recognition System for
Home-Service Robot by Integrating SURF and BRISK,” in Proc.
ICSSE, IEEE Xplore, DOI: 10.1109/ICSSE.2014.6887948, Jul. 2014,
Hungary.

Alle@Home.pdf
=============
s a first approach, we have integrated the package
find object 4 which recognizes objects in 2D using SURF descriptors and by means
of the depth information provided by the RGBD sensor, the 3D centroid on the
optical camera frame is computed [7]. Afterwards, the centroid is transformed
into the base coordinate system, whose origin is defined as the centre of the bot-
tom of the robot. We define the object pose by the centroid and a fixed rotation
angle, see Fig. 3.a. Our next approach includes the integration of 3D descriptors
as Fast Point Feature Histograms (FPFH) and deep learning techniques such as
tensorflow

7. Labbé, M. Find-Object. http://introlab.github.io/find-object, 2011. ac-
cessed 2017-03-10.

ASIL-TUT.pdf (HSR)
============
For general ob-
ject recognition, we have developed YOLO[19]-based object recognition system.
For specific object recognition such as particular bottles and cans, we have devel-
oped a method based on image matching[20] using several features, such as Scale
Invariant Feature Transform (SIFT) and Binary Robust Independent Elemen-
tary Features (BRIEF).

19 Yolo: Real-time object detection. https://pjreddie.com/darknet/yolo/.
20 find object 2d - ros wiki. http://wiki.ros.org/find_object_2d.

AUPAIR.PDF
==========
YOLO Ob-ject Detection (You Only Look Once: Unified, Real-Time Object Detection) [1]
(https://pjreddie.com/darknet/yolo/)

Imagenet Object Recognition (https://www.tensorflow.org/tutorials/image recognition)

BahiaRT.pdf (no information)
BIT@home.pdf (no information)
CIT Students @Home.pdf (no information)
CMCopper.pdf (no information)

Duckers.pdf (HSR)
===========
You Only Look Once (YOLO) 4 is used as object recognition algorithm

eR@sers2017.pdf (HSR)
[... 441 lines cut ...]
WrightEagle.pdf
===============
First the largest horizontal plane is extracted from point cloud using Point
Cloud Library (PCL)[11]. Then point clouds above the plane are clustered into
different pieces as candidates of objects. To improve the detection performance
and reduce false positive rate, clusters with abnormal sizes are filtered
out.Three kinds of features are used in this system: gradients feature in
object contours, HSV color histogram, and SURF feature. First, we use
LINEMOD[8] to detect the contours of object candidates in RGB-D images with
multi-modal templates. Matching based on this feature is unreliable, since
objects with similar shapes have similar contour gradient features. Thus HSV
color histograms are used to reduce false recognitions. After that SURF feature
matching is applied to the rest of the candidates to get the final recognition
results. We also use a joint model[13] to deal with occlusion and small objects
when detect objects. The joint model integrates scene classification, object
and visual phrase detection, as well as scene parsing together. By encoding
them into a Conditional Random Field model, occlusion and small objects could
be solved jointly.

[11] R. B. Rusu and S. Cousins. 3d is here: Point cloud library (pcl). In Pro-
ceedings of the IEEE International Conference on Robotics and Automation,
2011.
[13] I. Ulrich and J. Borenstein. People detection in rgb-d data. In Proceedings
of the 2011 IEEE/RSJ International Conference on Intelligent Robots and
Systems, page 38383843, 2011.

LoyVanBeek, 2017-10-17:
Popular methods/systems for classification: 
- SIFT, SURF
- YOLO
- GoogleNet
- find_object_2d/3d ROS packages
- Comparing H-S histograms
- ORK
- Google API
- Watson API

**Best team**: Homer@UniKoblenz, using SURF with some additional smart (sounding) tricks.

LoyVanBeek, 2017-10-17:
Pretrained nets: https://github.com/fchollet/keras/tree/master/keras/applications

LoyVanBeek, 2017-10-17:
Another suspicion is that the recognition is spoiled by bad segmentation. 
So, how can segmentation be improved? Segmentation from scratch on RGBD-images directly is a waste, because we already do background subtraction in ED with the world model. 

But the edges of the segmentation might be noisy and could thus be improved. We could use this asd a base for so-called 'weak supervision' or as a 'seed' to start the segmentation from. 

For [semantic segmentation](https://github.com/kjw0612/awesome-deep-vision#semantic-segmentation), the competition to look at is based on the [PASCAL VOC2012 dataset](http://host.robots.ox.ac.uk:8080/leaderboard/displaylb.php?challengeid=11&compid=5)

alberth, 2017-10-18:
I don't know how current ED is perceived, but if you ever want to get rid of the pre-baked world that Amigo needs (as I understood), relying on ED world model subtraction would become a PITA.

LoyVanBeek, 2017-11-04:
Cloud based object recognition services:
- [Google Vision API](https://cloud.google.com/vision/) (custom labels: :x: )
- [Microsoft Custom Vision (preview)](https://www.customvision.ai/) (custom labels: :white_check_mark: )
- [Microsoft Cognitive toolkit](https://azure.microsoft.com/en-us/services/cognitive-services/computer-vision/) (custom labels: :x: )
- [IBM Watson ](https://www.ibm.com/watson/services/visual-recognition/) (custom labels: :white_check_mark: )

LoyVanBeek, 2017-11-04:
Retraining DetectNet: https://github.com/NVIDIA/DIGITS/blob/master/examples/object-detection/README.md#model-creation

[DetectNet is derived from GoogLeNet](https://github.com/NVIDIA/DIGITS/blob/master/examples/object-detection/README.md#detectnet), which is also based on Inception_v3 which we use now. Retraining DetectNet should thus be possible.

reinzor, 2017-11-04:
> DIGITS/blob/master/examples/object-detection/README.md#model-creation

What were the conclusions of today\?

-Rein

LoyVanBeek, 2017-11-06:
@reinzor Not many conclusions (well, that rqt segfaults on my laptop) but mostly following tutorials and general study into this topic. 

People have been tweaking the threshold of putting a classification in the report of the Storing groceries challenge. A metric useful for tuning this could be the Precision-Recall curve:

- http://scikit-learn.org/stable/auto_examples/model_selection/plot_precision_recall.html#sphx-glr-auto-examples-model-selection-plot-precision-recall-py 
- http://scikit-learn.org/stable/modules/model_evaluation.html#precision-recall-f-measure-metrics
- http://scikit-learn.org/stable/modules/generated/sklearn.metrics.precision_recall_curve.html

LoyVanBeek, 2017-11-09:
Another interesting metric:
http://scikit-learn.org/stable/modules/generated/sklearn.metrics.roc_auc_score.html#sklearn.metrics.roc_auc_score

LoyVanBeek, 2017-11-12:
I trained a network on ~/MEGA/data/robotics_testlabs/training_data_Josja/training

I tested the network on our test_data in ```~/MEGA/data/robotics_testlabs/training_data_Josja/test_data```: 
[image]. 

and then again with the same data cropped to 90%, to naively get rid of a bit of background:
[image]

**Naively cropping makes classification much worse.** A good segmentation is important.

LoyVanBeek, 2017-11-14:
TODO:
- [ ] Get lots of images via "amigo, inspect the cabinet/bookcase/dinnertable" and put those in MEGA
- [ ] Annotate those with a large margin around the actual object, so we can test with varying margin down to smaller than the actual object
- [ ] Decide if classification is the problem or that better segmentation will help most

LoyVanBeek, 2017-11-14:
Trained on 6 classes with 60 images each. 1000 train steps, batch size of 100. 

[image]

LoyVanBeek, 2017-12-09:
More questions:

- [ ] Does using other team's data help or not?
- [ ] How many steps to train with?
- [ ] What batch size to train with?

LoyVanBeek, 2017-12-19:
More things to try out:
- [ ] Different pre-trained networks (See https://keras.io/applications/ for models and weights)
   - [ ] MobileNet
   - [ ] VGG16
   - [ ] VGG19
   - [ ] Xception
   - [ ] ResNet50

MatthijsBurgh, 2018-01-03:
I trained mobilenet, after I synced the tf_retrain script. Evaluation and classification isn't possible yet. Because the input_tensor "Cast" isn't in the main graph in Mobilenet. I don't know why this is and what the solution is. But using "input" as the input_tensor isn't working, because then the sizes are incorrect.
Probably we need to do JPG decoding in the script?

Network graphs:
### Inception
[image]
### Mobilenet
[image]

blumenkindC3, 2018-01-05:
I read a Paper about them. I will be back at the 7th, then I can send it
and give u an overview about accuracy and spend.

LoyVanBeek, 2018-01-09:
Judging by https://hackernoon.com/creating-insanely-fast-image-classifiers-with-mobilenet-in-tensorflow-f030ce0a2991, I'd  say that indeed the image does need to be resized:
```
python3 tensorflow/examples/label_image/label_image.py \ 
  --graph=/tmp/mobilenet_0.50_192.pb \
  --labels=/tmp/output_labels.txt \
  --image=/home/harvitronix/ml/blogs/road-not-road/test-image.jpg \
  --input_layer=input \
  --output_layer=final_result \
  --input_mean=128 \
  --input_std=128 \
  --input_width=192 \
  --input_height=192
```

closed 2018-10-17 by MatthijsBurgh