tue-robotics/tue_robocup #402: stable_grasping

hvrooy, 2017-09-26:
Improving grasping and manipulation of the robot will be beneficial for the results of demo's and tournaments. Improvements areas are:

## General
- [ ] research current publications and papers on the subject and set our goals
[Stable_Grasping_under_pose_uncertainty.pdf](https://github.com/tue-robotics/tue_robocup/files/1334613/Stable_Grasping_under_pose_uncertainty.pdf) is a good starting point.

## Hardware
- [ ] improve the design of the current grippers
- [ ] improve force feedback of the current grippers
- [ ] adding a few (analogue) sensors to the gripper to detect objects in range of the gripper
- [ ] adding a lot of small pressure sensitive sensors on the fingers to obtain 'grasp-signatures'  for different objects
## Software
- [ ] Improving segmentation in ED
- [ ] have sensor information influence the trajectory of the arm (adaptive) to result in a stable grasp
- [ ] apply machine learning to grasping

hvrooy, 2017-10-01:
www.kiwi-electronics.nl has info on the ADAFRUIT VL6180X TIME OF FLIGHT DISTANCE RANGING SENSOR (VL6180) and several others. Maybe worth an experiment.

hvrooy, 2017-10-16:
To generate a signal that can be used to influence the path of the hand we can experiment with small-range sensors (see www.kiwi-electronics.nl) that are based on:
- [ ] time of flight (laser) - VL6180 - 5..200mm
- [ ] infrared light 
- [ ] ultrasound 
- [ ] red light (fiber optics)
- [ ] contact pressure - small sensor pads on the fingers of the gripper

Space near the grippers is limited, so the sensors should be as small as possible, or at least the sensing element(s) should be. Signal conditioning and amplification could be done elsewhere. 
It should be checked if two contactless sensors positioned at an angle and close enough near the gripper in order to 'see' the object to be grasped, can generate a useful signal to influence the grasping XY-path of the arm/gripper so that grasping is performed better. (Or do we need at least three sensors if want to influence the Z-axis as well?)

Of course, the way these signals are processed in the control software is also something to be discussed.

jlunenburg, 2017-10-27:
Using two sensors might still be difficult. Ideally we'd be able to measure in a plane. Don't know if there are any light screens in the desired dimensions.

hvrooy, 2018-01-11:
Components for first experiment: 2 TOF sensors with different ranges (5..200 and 20..2000 mm), 1 Arduino R0 Pro (this has a DAC on board). Sensor(s) are interfaced with I2C bus. Resulting analogue signal will be injected in the PERA board in the arm of AMIGO for futher processing and control.
Noise may be an issue close to the DC motors in the arm, so the I2C wires should be shielded and as short as possible. The same goes for the analogue signal. (Voltage 3V3, NOT 5V)

hvrooy, 2018-01-23:
First experimental setup attached to AMIGO's left arm is ready for testing. 
VL6180X TOF sensor [0..180 mm] --> Arduino M0 Pro --> DAC output (0...3.3 Volt) -> TUe-ES020 sensor input in AMIGO's lower arm.

MaxBaeten, 2018-01-23:
I’ll be there in a sec :)

hvrooy, 2018-01-23:
Initial test was successfull. Lower and upper bounds of DAC output of Arduino will be used for detecting error conditions. No power for Arduino (low value), Out of range (max value). Valid data in between.
Changes to sensor casing design:
- Place sensor 1 cm backwards, and make provisions to be able to adjust the tilt. (fine tuning)
- Incorporate the coverplate of the gripper in the casing design, if possible.
- Make provisions to add two more sensors in the future.
- Locate a stable 5V (or 3V3) power supply for the Arduino in the arm of AMIGO.

closed 2018-12-01 by MatthijsBurgh