tue-robotics/tue_robocup #446: Spindle

MatthijsBurgh, 2017-10-23:
During homing, the spindle went up and hit the end or something and made a weird collision noise. Dashboard said homing was correct, but was not. I homed again. It was able to go up for a few cm's and went down to final position.

hvrooy, 2017-10-23:
Strange behaviour. I am beginning to think the homing software is at fault here. There a two sensors that detect the upper and upper-upper limit, and both seem to be ignored somehow. Maybe they should  be more 'hard wired' in the safety circuit of the spindle motor. (I am not quite sure how to do that yet, though). To be continued.

MatthijsBurgh, 2018-01-13:
@MaxBaeten you did some software fixes. Are these related to this issue?

MaxBaeten, 2018-01-13:
No i did not change anything related to spindle homing. But i’ve never seen this issue myself

MatthijsBurgh, 2018-01-13:
Ok, lets close it for now. I haven't seen it happen for more than a month.

closed 2018-01-13 by MatthijsBurgh