tue-robotics/tue_robocup #459: [SPR/robot_skills] Implement openface and skybiometry in head

MatthijsBurgh, 2017-10-24:
Openface is already implemented. I don't know if the api in the SPR and robot_skills the same are. So check if this is usable and also implement skybiometry in robot_skills

LoyVanBeek, 2017-10-25:
We used to have a 'perception' RobotPart fot this.
Putting perception stuff in Head, which also has a bunch of movement-related stuff seems weird

MatthijsBurgh, 2017-10-25:
There is already a lot of it in there, so maybe you want to seperate it. But otherwise this is the way to go in my opinion.

LoyVanBeek, 2017-10-27:
@jlunenburg opinions?

Rayman, 2017-10-27:
Perception is good for me, then the head has only actuation in it.

LoyVanBeek, 2017-10-27:
That would make sense to me as well. We also have SSL and HMI separated.

jlunenburg, 2017-10-27:
Sounds sensible. Suppose you have a camera that is not mounted on the head.... 

Think about this there are more things that could be improved ;)

PR #493 (2017-11-14): Refactor perception
Merge together with: tue-robotics/amigo_bringup#81 and tue-robotics/robot_launch_files#20
fixes #459 and fixes #354
closed 2017-11-14 by MatthijsBurgh