tue-robotics/tue_robocup #461: Timing issue: PCs are not in sync after startup

jlunenburg, 2017-10-25:


Timple, 2017-10-26:
For a next time, could anybody please post the output of `chronyc tracking` on every amigo pc?

MatthijsBurgh, 2017-10-26:
# Boot without internet
## AMIGO1
```
506 Cannot talk to daemon
```
## AMIGO2
```
Reference ID    : 127.127.1.1 ()
Stratum         : 10
Ref time (UTC)  : Thu Oct 26 13:10:23 2017
System time     : 0.000000001 seconds fast of NTP time
Last offset     : +0.000000000 seconds
RMS offset      : 0.000000000 seconds
Frequency       : 4.744 ppm fast
Residual freq   : +0.000 ppm
Skew            : 0.000 ppm
Root delay      : 0.000000 seconds
Root dispersion : 0.000001 seconds
Update interval : 0.0 seconds
Leap status     : Not synchronised
```
# Connect router to internet
## AMIGO1
```
Reference ID    : 127.127.1.1 ()
Stratum         : 8
Ref time (UTC)  : Thu Oct 26 13:08:14 2017
System time     : 0.000000000 seconds fast of NTP time
Last offset     : +0.000000000 seconds
RMS offset      : 0.000000000 seconds
Frequency       : 7.811 ppm fast
Residual freq   : +0.000 ppm
Skew            : 0.000 ppm
Root delay      : 0.000000 seconds
Root dispersion : 0.000001 seconds
Update interval : 0.0 seconds
Leap status     : Not synchronised
```
few seconds later
```
Reference ID    : 83.162.149.224 (schnitzel.team)
Stratum         : 3
Ref time (UTC)  : Thu Oct 26 13:05:52 2017
System time     : 139.909072876 seconds fast of NTP time
Last offset     : +0.000089441 seconds
RMS offset      : 0.000089441 seconds
Frequency       : 7.811 ppm fast
Residual freq   : -17.963 ppm
Skew            : 0.203 ppm
Root delay      : 0.009966 seconds
Root dispersion : 0.010906 seconds
Update interval : 2.2 seconds
Leap status     : Normal
```
## AMIGO2
```
Reference ID    : 127.127.1.1 ()
Stratum         : 10
Ref time (UTC)  : Thu Oct 26 13:11:19 2017
System time     : 0.000000002 seconds fast of NTP time
Last offset     : +0.000000000 seconds
RMS offset      : 0.000000000 seconds
Frequency       : 4.744 ppm fast
Residual freq   : +0.000 ppm
Skew            : 0.000 ppm
Root delay      : 0.000000 seconds
Root dispersion : 0.000001 seconds
Update interval : 0.0 seconds
Leap status     : Not synchronised
```
few seconds later
```
Reference ID    : 192.168.44.91 (amigo1)
Stratum         : 4
Ref time (UTC)  : Thu Oct 26 13:09:10 2017
System time     : 0.000005448 seconds slow of NTP time
Last offset     : -0.000006729 seconds
RMS offset      : 7.565318108 seconds
Frequency       : 6.353 ppm fast
Residual freq   : -0.008 ppm
Skew            : 0.363 ppm
Root delay      : 0.010117 seconds
Root dispersion : 0.005443 seconds
Update interval : 32.2 seconds
Leap status     : Normal
```
# Final times
AMIGO1
```
Thu Oct 26 15:13:07 CEST 2017
```
AMIGO2
```
Thu Oct 26 15:10:53 CEST 2017
```
# After restarting chrony service on AMIGO2
When observing for a longer time. AMIGO1 is reducing its ahead of NTP time slowly. But both the Ref time in chrony as date are not converging.
Ref time is ahead on AMIGO2, while date is ahead on AMIGO1.

Hopefully you have enough information

Timple, 2017-10-27:
Yup, I remember the issue now.

### issue
- amigo1 does not step to the time upon connecting to the internet (this is desired behavior, we don't want random steps in the midst of a demo).
However sinds it knows the correct time, it does feed this to amigo2. 
- amigo2 then steps to this time (being in sync with amigo1 is more important than anything).

### Potential solution
amigo2 should broadcast its own time, not the 'best' time.
This cannot be configured in chrony (afaik). Maybe setup a seperate time broadcaster on amigo1 (chrony/ntp) which does not have knowledge of the internet time.

Rayman, 2017-10-27:
Do we host an NTP server on amigo1?

LoyVanBeek, 2017-10-27:
Amigo is a Chrony server IIRC. 

For Amigo, being in sync is more important that being on the actual time, so I don't see how broadcasting time from Amigo2 helps. 

But: why isn't the time reported from the hardware clock (in BIOS etc) not correct/ Is the battery on the motherboard empty? Systemd syncs the NTP-received time to the hardware clock on shutdown?

jlunenburg, 2017-10-27:
I guess it's not set on shutdown. Might indeed be an easier fix, although you have to be really sure it's correct. I assume the clocks would be off further if the batteries are really empty. Easy enough to check this using `hwclock` though

LoyVanBeek, 2017-10-27:
I can imagine them not being set when there's a sudden power cut?

Timple, 2017-10-28:
> Do we host an NTP server on amigo1?

Yes, Chrony does this

> so I don't see how broadcasting time from Amigo2 helps

Who's talking about broadcasting *from* Amigo2?

> But: why isn't the time reported from the hardware clock (in BIOS etc) not correct/ Is the battery on the motherboard empty?

Who says it is not? But anyway irrelevant I think, after a weekend this should work as well, and amigo1 and amigo2 hardware clocks will differ too much after a couple of days.

> I can imagine them not being set when there's a sudden power cut?

As far as I can recall the hardware clock is written periodically exactly for this reason. But again, what would any of this solve? When amigo1 finds internet is synchronises, passes the right internet time to amigo2 and amigo2 jumps to this time while amigo1 does not. How does writing hardware clocks prevent any of this?

Timple, 2017-11-04:
Should be solved once https://github.com/tue-robotics/tue-env/pull/194 and https://github.com/tue-robotics/amigo_bringup/pull/78 are merged.

Lets monitor for a bit.

Rayman, 2017-11-21:
Test plan:
- start amigo without internet
- see if pc's are in sync

@MatthijsBurgh will look at this

MatthijsBurgh, 2017-12-05:
I already tested it a week ago. The PC were not in sync with my laptop, which is correct in case of no internet. All AMIGO pc's were in sync. So seems to be working. Closing issue.

closed 2017-12-05 by MatthijsBurgh