tue-robotics/tue_robocup #475: Add status publisher to action server

jlunenburg, 2017-10-31:


rokusottervanger, 2017-11-04:
Proposition: action server publishes its status on an action_server/status [action_server_msgs/status] topic on change (latched, so you can subscribe and still get the latest status).

action_server_msgs/status:
```
time stamp
string description
```
What should this description contain?

LoyVanBeek, 2017-11-04:
- Which command are we processing
- What action is running

Can't this be a standard Diagnostics message?

Rayman, 2017-11-04:
@rokusottervanger and me discussed this. There is an actionlib interface so the GUI already knows when it's busy and when it's finished. This has to be implement only on the GUI side.

jlunenburg, 2017-11-28:
In general, functionality is missing in the current demo UI

jlunenburg, 2017-11-28:
@rokusottervanger  When I call a cancel on the roslibjs actionclient, I can see the cancel message published over the `/amigo/action_server/task/cancel topic`.

closed 2017-11-28 by jlunenburg