tue-robotics/tue_robocup #480: Threshold on classification probability

rokusottervanger, 2017-10-31:
The classified objects with low probability should be labeled unknown, because this does not result in reduced score, whereas false positives result in negative score. (But check the rulebook on this rule).

jlunenburg, 2017-11-01:
I put the threshold back from 0.01 to 0.3 in the merge request concerning this issue (I assigned @rokusottervanger as reviewer), the previous value. We have been experimenting with this value in Japan to see how we could get the best score, but I guess that when we selected 0.01 we were still assuming that the objects on the table would also count and that all classifications had extremely low probability. Hopefully this is better now.

@JosjaG might also be interesting for you!

MatthijsBurgh, 2018-01-09:
Related to #525