tue-robotics/tue_robocup #481: action list is not cleared

rokusottervanger, 2017-10-31:
After executing (and failing) task "find the person near the cabinet and answer a question", the robot tries to answer a question at the start of the next given command.

rokusottervanger, 2017-11-14:
To reproduce:
```
Start simulator 
start challenge-gpsr
amigo-hear amigo
amigo-hear <some task that crashes> and answer a question
amigo-hear yes
```
Amigo will start executing the command. After it fails and Amigo returns to the meeting point, try a new task (something else than 'answer a question':
```
amigo-hear go to the hallway
```
Amigo will first try to 'answer a question'

closed 2018-01-02 by timassman

timassman, 2018-01-02:
Fixed in https://github.com/tue-robotics/action_server/pull/89