tue-robotics/tue_robocup #490: Allow preemption of demo presentation state

rokusottervanger, 2017-11-07:
We want to be able to cancel the presentation while it is going. Currently, this cannot be done. Smach has a way of doing this. Tutorial can be found [here](http://wiki.ros.org/smach/Tutorials/State%20Preemption%20Implementation).

You do want to avoid generating a lot of duplicate code by checking if preempt is requested like this:
```
...
self.robot.speech.speak(self.trans.HI_MY_NAME_IS.format(self.robot.robot_name), language=self.language, voice=self.voice, block=True)
if self.preempt_requested():
    return 'preempted'

self.robot.speech.speak(self.trans.IM_A_SERVICE_ROBOT, language=self.language, voice=self.voice, block=True)
if self.preempt_requested():
    return 'preempted'

self.robot.speech.speak(self.trans.PURPOSE, language=self.language, voice=self.voice, block=True)
if self.preempt_requested():
    return 'preempted'
...
```

You could use`functools.partial` (doc [here](https://docs.python.org/2/library/functools.html)) to make a list of all functions that should be called in the state machine, and then loop through this list. In the loop, you can call the next element in the list (the next function) and check if preemption is requested.

Rayman, 2017-11-07:
I think your states are then too big. If you have multiple small states this wouldn't be a problem.

rokusottervanger, 2017-11-07:
@Rayman I agree that this is state behavior and should, strictly speaking, be modeled as a state machine. However, this is a completely linear behavior, so scripting is more convenient, much less verbose and more readable than instantiating a Smach state for every function call.

alberth, 2017-11-08:
There is still a lot of repetition in your scripting example, it seems. You can't write a function that takes the list of functions to call, which then generates a state machine for it? (or a set of states, for that matter). That makes it simpler to wevae these sequences into other parts of the state machine where you need to be more careful.

Alternatively, if you make "not preempted" a condition for executing the action in each state, you don't need the "preempted" exit any more (once preempted becomes true, you blast through the remaining states in a fraction of a second as you skip all actions), and it becomes a simple sequence. Would that help?

rokusottervanger, 2017-11-08:
In my suggested solution there is no repitition. I didn't write an explicit example for that because this is a good exercise for @KevinDang74 . You could automatically generate a Smach state for each function call, but what is the added value of that if these states are always executed sequentially anyway?

The suggestion for checking 
```
if not self.preempt_requested():
    robot.skill.do_something()
```
indeed reduces the number of lines of code by 1, but then we still need to check for 'preempt_requested' to return the correct outcome ('preempted') as the governing state machine (or any other caller of the execute() method) should be notified that the state did not succeed normally, but was interrupted by a preempt request.

All in all, I could have made three implementations of this in the time we spent discussing this topic. However, my goal was to give Kevin a simple assignment to improve upon his Smach skills. Why don't we just let him figure out a nice solution and then review his PR ;)

KevinDang74, 2017-11-09:
Thanks for the comments and suggestions, I will work on this as soon as possible!

closed 2017-11-28 by KevinDang74

MatthijsBurgh, 2017-11-28:
Fixed by #518