tue-robotics/tue_robocup #496: Improve WM creation

MatthijsBurgh, 2017-11-14:
In the ed_robocup plugin the type is inhered from parent type. This is unwanted an inconsistent with loading a WM in the default way. Because the inheriting is done in the ed model_loader, I don't want to change that one. A dirty fix is needed. This is done in tue-robotics/ed_robocup#5
A second fix is needed in ED, tue-robotics/ed#56. During processing the update requests, the type is retrieved from data of the update request. This is double because this is also done in the model_loader. Plus the plugin should have the authority to set the type, independent of the data. Therefore this should be removed.

Desired output of dump_ed:
```
- type: "table_120x80x76"
      id: "dinner_table"
      pose:
          x: 0.4
          y: 1.92
          z: 0.0
```
Current output:
```
- type: "table"
      id: "dinner_table"
      pose:
          x: 0.4
          y: 1.92
          z: 0.0
```

MatthijsBurgh, 2017-11-14:
@reinzor this explains the PR's in ED and ed_robocup

alberth, 2017-11-15:
Dirty fixes only create much more work, not less.

MatthijsBurgh, 2017-11-15:
Not for this one :p

alberth, 2017-11-15:
Of course it also holds for this one.

Unless you aim to remove ED, there will be a next person that improves it somewhen. That person has to understand your addition is a work-around, understand why it was added, understand what the proper fix is, do the proper fix, remove the work around, and check that it works.

This may happen long after you're gone (ie your knowledge is not available any more), or the original reason for adding the hack may have become obsolete.

closed 2020-03-31 by LarsJanssenTUe